iNavFlight/inav #9311: iNav 6.1 fixed wing cruise navPos[2] jumps

helicocrasher, 2023-09-24:
I have observed two strange things with iNav 6.1 on two of my flying wings. 

1. At some occasions in cruise mode the vario is showning constant sink of ~1m/s (wrongly) and the height is staying constant (as expected).
2. Somtimes in cruisemode the OSD shows a very brief jump in height +10 or 20 meters. The NavPos{2} in blackbox is also showing these jumps.  Baro and GPs height do not have this jumps 
As a consequnce the plane is immedately throttling down to correct for the perceived height error. 

I have recently puled 11 blackbox logs from one of my wings. 9 do not show this behavior 2 have it. 
I do not recall to have made changes t the settings. 
So I have no idea how to reproduce or provoke the behavior.

## Steps to Reproduce

1.
3.
4.
5.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

breadoven, 2023-09-25:
It would be useful if you could provide the full log file.

Feldsalat, 2023-09-27:
I have jumps like that too. Not always and on different planes. Just like here:
[https://youtu.be/Oqidg0h63Gc](url)
I don't have any logs.

breadoven, 2023-09-27:
Something doesn't look right there with the vertical velocity. It's constantly negative over a circa 30s period but vertical position remains roughly constant as does the Baro altitude. Almost looks like some sort of error correction kicks in bumping up the vertical position by the height that would have been lost of the vertical velocity was correct. Accelerations don't look excessive so not obvious what's causing it unless it's a GPS issue. Check the GPS heights.

helicocrasher, 2023-09-27:
@breadoven thanks 
Yes vertical velocity is weird. 
I have put the OSD video here https://youtu.be/iYJT-RL6dj0
The sequence with the 4 bumps in the middle of the graph.
Then below a new graph with the GPS height added. 
Someway the GPS height looks like scaled up ... 

[image]

helicocrasher, 2023-09-27:
More Flight height graphs 

[image]

[image]

b14ckyy, 2023-09-28:
This was fixed in 6.1.1 and was the main reason for that patch. See https://github.com/iNavFlight/inav/pull/9111

helicocrasher, 2023-10-01:
More Flight height graphs 

[image]

closed 2023-10-01 by helicocrasher

helicocrasher, 2023-10-01:
> This was fixed in 6.1.1 and was the main reason for that patch. See #9111

Uh so simple! I missed that release.
@b14ckyy  Thanks
I will upgrade my planes now.