iNavFlight/inav #9315: ref https://www.ga.gov.au/__data/assets/pdf_file/0011/123320/SBAS-STN-0001_02_SDD-OS.pdf

stronnag, 2023-09-26:
ref https://www.ga.gov.au/__data/assets/pdf_file/0011/123320/SBAS-STN-0001_02_SDD-OS.pdf
     https://www.amsa.gov.au/safety-navigation/navigation-systems/satellite-based-augmentation-system-sbas

Can we add SouthPAN SBAS for users in this Australia and New Zealand region please?  
Currently, if SBAS is set to **auto** in 6.1.1 the GPS struggles to get a position.  
If disabled we get a much lower position accuracy and longer sat acquisition times.

thanks

_Originally posted by @SpidaTuSefen in https://github.com/iNavFlight/inav/discussions/9313_

stronnag, 2023-09-26:
About time too!

SpidaTuSefen, 2023-09-27:
LOL, I'm just so happy it's here and can't wait to use it mate - the first SBAS sat in the Southern Hemisphere!😁

mmosca, 2023-09-27:
In auto mode, the gps will accept any SBAS system it sees, so it should automatically pick up SouthPAN.

I will add it, but I am skeptical if it will help much. You guys have a single PRN.

mmosca, 2023-09-29:
Code has been merged.

closed 2023-09-29 by mmosca