iNavFlight/inav #9328: [6.1.1] fixed wing, iterm to zero on fast rolls?

0crap, 2023-09-30:
## Current Behavior
On "fast" rolls the bounce on center stick (when you stop the roll) is large and easily seen with FPV flights.
`set fw_iterm_limit_stick_position =  0.125` is already in place and limits the I term during a roll.

When doing the same roll in manual mode, there is very little bounce to be seen when you stop the roll. 
The Blackbox log shows why, **iterm is zero when in manual mode**.

[image]

[image]

## Expected behavior
No bounce when you stop a roll in ACRO mode.

## Suggested solution(s)
In ACRO, create a setting that can set iterm to zero during fast moves.

# version
# INAV/MATEKH743 6.1.1 Jun 13 2023 / 11:52:23 (42ced253)
# GCC-10.2.1 20201103 (release)

Jetrell, 2023-10-01:
@0crap This has certainly been an issue for a while.
The problem with `fw_iterm_limit_stick_position`. Is that it only prevents the i-term from growing.. It doesn't reset it back to zero. Which still leaves some accumulated i-term, that has to unwind at center stick.
I think it would be better if it cleared the i-term error, when the stick released speed (back to center), exceeds the PID loop time when it sees the i-term shrinking.

0crap, 2023-10-02:
Indeed an issue for quite a while.
And frustrating everytime when you try to tune a new airplane.
I sure hope the time is right to fix this one.
(After the AHI fix this is now my number one issue to make INAV perfect for planks.)

A proper fixed wing iterm relax on roll should fix this.
Pitch is not an issue because a looping is a relatively slow move.
Roll stops however is an pain in the butt...

0crap, 2024-12-10:
[Housekeeping] This one should finally be closed by https://github.com/iNavFlight/inav/pull/9905
Which we will test upcoming flying season.