iNavFlight/inav #9331: Support FLYWOOF405NANO Target

Lethal-Paw, 2023-10-01:
## Current Behavior
The Flywoo goku GN405 Nano, Flywoo goku hex GN405 and Flywoo goku versatile F405 PRO are currently the smallest FCs on the Marked tht support Inav. Thismakes them very appealing vor very small Long range quads and micro wings. Sadly they do not have an official INAV Target so we have to pullthe software from flywoos Help desk. Problem is: The firmware has not been updatet forover a year. https://flywoo.tawk.help/article/firmware-inav

## Desired Behavior
Add the Targets for these three FC Boards. As far as I understand Flywoos Target description, they all share the FLYWOOF405NANO target.

## Suggested Solution
Add FlywooF405nano target

## Who does this impact? Who is this for?
- Everyone flying Flywoos Ultralight Long range copters
- Everyone using super small 16x16 and 20x20 flight contollers for their models
- Everyone who can't fit a Full sized FC in their Model

Martin-MK-UK, 2023-10-04:
This would be really helpful. I'm having to use beta flight for my multirota and I want to use INAV

dz0ny, 2023-10-05:
See https://github.com/iNavFlight/inav/pull/8606

b14ckyy, 2024-03-22:
Until Flywoo starts to care about INAV and follows the policy, this will stay an inofficial target.

closed 2024-03-22 by b14ckyy