iNavFlight/inav #9368: ESC output Standard 50 Hz framrate

wolfffgang, 2023-10-13:
- FC Board name and vendor: Speedybee F405 wing
- INAV version string: 2023-10-13 @ 12:51:13 -- Running - OS: Windows, Chrome: 98.0.4758.82, Configurator: 6.1.0
- 
 Output setup for Standard ESC produces a frame rate of 400 Hz.
In the wing applications the framerate of 90% populat ESC use as Standard 50 Hz framerate.
I tested with Speedybee F405 wing standard esc from robbe

DzikuVx, 2023-10-27:
All modern ESC support this PWM rate. And by modern I mean even 8 years old

closed 2023-10-27 by DzikuVx

dakopian, 2024-04-20:
I have the same problem with my Mini Apprentice S. It doesn't work with a 400Hz refresh rate. Please add an option to slow that down.

sensei-hacker, 2024-04-20:
> I have the same problem with my Mini Apprentice S. It doesn't work with a 400Hz refresh rate. Please add an option to slow that down.

If you have one that's actually incapable of 400 Hz, you can instead connect it as "servo".  In the mixer, just set that "servo" as stabilized throttle.  This is how we always did it before quads ("drones") were around, needing much higher speed ESCs.

When doing that, be sure to also always remember the old rule - any time the battery is attached, the prop can spin - so prop removed when you're working on the plane indoors!

dakopian, 2024-04-20:
Thanks! Actually, I already replaced the ESC with a BLHeli_S-based one. Given their affordability, there was no sense in messing with that old ESC. Everything works perfectly now.