iNavFlight/inav #9369: led management on SPEEDYBEE F405 v3

sgrizzi, 2023-10-13:
I am building a new quadcopter using the SPEEDYBEE F405 V3 stack, running INAV 6.1.1.
The quadcopter is using a 4S battery, an ELRS receiver, a BE880 GPS+magnetometer. The ESC drives 4x 2207 motors.
The sw setup is working fine so far. 

The SPEEDYBEE F405 V3 flight controller can also support a led strip connected to one of the two pad strips provided.
No problem to control this led strip using the INAV configurator.
The flight controller also sports 4 led strip outputs at the corner of the board, labeled D1..D4.
Is there any way to configure/drive these 4 led outputs, using the INAV configurator? or otherwise?
I could not find a solution to this problem, I hope someone in the group can give me a hint.
Best regards.

mmosca, 2023-10-14:
I am not familiar with the F405v2, but if it works in a similar way to how other speedybee flight controllers work, they will either duplciate the signal from the led pad, or be controlled by speedybee directly, with modes changing by pressing the boot button.

If you want to control if rom INAV, you may need to daisy chain all leds on the same port.

sgrizzi, 2023-10-18:
Apparently the led's on the FC corners are independent from one another - maybe directly controlled by F405 ports.
I don't have the schematics of the SPEEDYBEE F405 V3 and do not see any way to control these leds using INAV.
Indeed I can daisy chain "other" leds and connect them to the "led strip" output which can be controlled by INAV.

closed 2023-10-21 by sgrizzi