iNavFlight/inav #9371: Onboard Takeoff Push Button For Large Fixed Wings

gaaryz, 2023-10-15:
Current Behavior :

To take off in AutoLAUNCH mode, with the remote control on the ground, for a larger aircraft that requires a two-handed grip on the fuselage, the throttle is brought up to the maximum speed either by a throw or a shake. In my specific use case, I prefer using a shake, where I hold the aircraft until the engine starts and reaches its full throttle. Then, I launch the fixed-wing.

Desired Behavior :

When AutoLAUNCH is active, instead of a throw or a shake, you could double-press a button located on the fuselage, near one of the two hands, to initiate the AUTO takeoff. The Autopilot will then start the engine and bring it to full throttle. Afterward, you launch the fixed-wing.

If, for any reason, you need to cancel the takeoff while the engine is running, you can press the button once to DISARM the aircraft. The engine will shut down, and the takeoff will be canceled.

Suggested Solution :

Add an option in the Plane configuration as well as in the port selection bar to configure the use of a button for takeoff.

On the hardware side, the button would be connected as a PullUP/PullDOWN on an RX pin of a UART on the Flight Controller (FC), using 3.3V to set it to the desired state.

On the code side, the used UART would interpret the button's action as a boolean. This information could be incorporated into the AutoLAUNCH mode logic, replacing the previously calculated acceleration trigger for the start of takeoff.

 Who does this impact? Who is this for?

This solution would allow users of any kind of fixed wings to configure and use a button for initiating takeoff, making the process more versatile, user-friendly and, above all, safer.
I myself dream of this option to be able to cut the throttle on the day I need to abort a takeoff. With a 17-inch propeller spinning at full speed on my 5kg, 3-meter wingspan motorglider, it could be a real hassle XD

Love You INAV ♡
## Additional context

mmosca, 2023-10-15:
while the sugestion is good, the hardware implementation may not fit the current inav resource model well. 
An alternative would be to have a small mcu handle the button and send a msp message to inav to start auto launch.

MrD-RC, 2023-10-15:
I don’t think this is needed when the option exists to use autolaunch with manual throttle.

RomanLut, 2023-10-16:
I would suggest not implementing rarely used features into inav but enhance programming tab instead. Add a logic input to sense logic level on UART's RX pin, and a operation to post event with id (in this case f.e. id = 0 - launch detected), and let people be creative in programming tab.

MrD-RC, 2023-10-16:
In all honesty. I think the current setup is just fine for large aircraft with the transmitter on the ground. This PR would help a little, by triggering idle throttle https://github.com/iNavFlight/inav/pull/8511

I misread and thought the large aircraft were on the ground (undercarriage). Hence the suggestion for manual throttle.

mmosca, 2023-10-16:
Having the ability to repurpose a uart into a general gpio may not be that simple as inav lacks the foundation to implement resource remapping and the pins would be in use by the uart already. 

It quite possibly would require a custom target. 

[]s,
Marcelo Bezerra ***@***.***>

RomanLut, 2023-10-16:
No need to repurpose UARTs. Just let reading logic level on UART's RX pin with IORead() operation regardless what UART is configured for. This would be enough to implement custom buttons and logic level inputs. UART can be confgured for LTM/MSP/whatwever just ot intitalize RX pin as input. Button has to be implemented as stronger pullup resistor + pushbutton as pulldown.

MrD-RC, 2023-10-16:
@mmosca there are plenty of targets that already have PINIO. But, as far as I know, at the moment it is only PINO. Having the input side accessible through the programming framework may be a good start. Not just for this, but many applications.

breadoven, 2023-10-16:
Actually the "point the nose at the sky" method of cutting the motor after launch detection should be possible now just using current programming logic, launch state + pitch attitude + throttle override set to 0, or maybe also channel override to disarm completely ?.

sensei-hacker, 2023-11-07:
This can be done without any changes to INAV, and without any buttons.  Just by using the features as designed.

The first half of this would be resolved by proper use of launch idle throttle and nav_fw_launch_idle_motor_delay.
The second half, changing your mind and disarming, could be done as @breadoven suggested without any changes to INAV.

Set nav_fw_launch_idle_motor_delay to perhaps 20,000 or whatever value you choose.
Engage autolaunch.  That gives you 20 seconds to pick up the plane.

After that 20 seconds (idle_motor_delay), the user wants it to go to approximately full throttle.
The throttle value before launch is set by nav_fw_launch_idle_thr
The "problem" only comes about because of the use of launch_throttle instead of nav_fw_launch_idle_thr.
If you want full throttle before launch, set nav_fw_launch_idle_thr and the problem goes away.

It throttles up after 20 seconds (or however long you choose), then just throw it. :)

If you decide you want to cancel the launch, breadoven's solution works.

b14ckyy, 2024-03-22:
I also think a button to trigger auto launch is overkill and unnecessary. 

We have a reliable launch detection for the throw
We have a adjustable timer for up to 60s to give plenty of time to prepare to throw
We have https://github.com/iNavFlight/inav/pull/8511 that should finally be merged for 8.0 and allows to trigger Idle throttle by just wagging the tail of the plane (that would be equivalent to the button press and needs no hardware and no target changes)

closed 2024-03-23 by b14ckyy