iNavFlight/inav #9374: MSP VTX OSD channel designator changes at arming

MrMap, 2023-10-15:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1. Start out with e.g. Raceband Ch5 Pwr 2, using the EDGETX ELRS VTX LUA, confirm OSD display correct CH: R5 : 2
2. Arm (Aux 1 / CH5)
3. Watch the OSD changes to CH: A1: 2
4. The VTX Channel has not changed, it will still be on R5

## Expected behavior

## Suggested solution(s)

## Additional context
I use HDZero goggles, all FW  22082023
HDZ Freestyle v1, current FW _22082023

Here is a short DVR video demonstrating the phenomena.
I have considered if this maybe is normal but I think it is not?

---

- FC Board name and vendor:
- # version
# INAV/SPEEDYBEEF405WING 7.0.0 Sep 11 2023 / 12:56:24 (3a2412e5)
# GCC-10.3.1 20210824 (release)

MrD-RC, 2023-10-15:
My friend had this. It’s an HDZero bug. Downgrading the HDZero firmware fixed the issue for him.

MartinHugh, 2023-10-25:
> My friend had this. It’s an HDZero bug. Downgrading the HDZero firmware fixed the issue for him. For some reason arming changes the channel.

Interesting. Could you please verify which firmware and which version.  I am assuming your friend is referring to the VTX firmware, and to do MSP-VTX it would have needed v1.4.0.  If he has rolled back to anything earlier MSP-VTX is not supported anyway. 

Any other details on exactly what was changed would be helpful.

MrD-RC, 2023-10-25:
All I know is that he tried the latest HDZero firmware (this would be around 3 or 4 weeks ago) and he had problems with the channel; as described here. He rolled back to the previous version, and it all worked fine. I don’t use HDZero myself. I’ll see if he knows the version number.

aerostarlegacy, 2024-03-11:
Hello, I think I am also having this issue. I have goggles and race V2 vtx on newest firmware. When using stick commands in osd to change channel or power the OSD reflects the new values, but the channel and power do not actually change. @MrD-RC  were you able to get that extra info from your friend? it may help me or others.
Thanks!

b14ckyy, 2024-03-23:
Will move this to discussion as it seems to be not a INAV issue.