iNavFlight/inav #9377: support for STM32F411

akhodeir, 2023-10-17:
I have some chinese boards around with STM32F411CEU6. I was thinking to use them for INAV for fixed wing plane.

I found some INAV flight controllers that support this MCU ex. mateksys F411-WSE and mateksys F411-WSE.

The MCU has "Flash = 512 Kbytes" and "SRAM 128 Kbytes". It sounds like it has limited capabilities. 

My questions : 

1- would there be INAV features that I lose due to the limited capabilities of the MCU ? or all features can fit inside it?
2- if I deisng my own flight controller, would it work stright forward with minor modifications ?

akhodeir, 2023-10-17:
Wow, This is big surprise to me. 
Thanks for letting me know.

what MCU is recommended?

YI-BOYANG, 2023-10-17:
Go for the F405 on the cheap and the more expensive F745 or H743.

akhodeir, 2023-10-17:
I will check if I they are available in my area. 
Is there any list of recommended MCU ?

YI-BOYANG, 2023-10-17:
Will different MCUS be disabled in some regions? I don't quite understand.

akhodeir, 2023-10-17:
In my area they don't sell MCU as component. 
it is only found on boards :)
so I have to search for NUCLEO board or whatever....

akhodeir, 2023-10-17:
anyway, you gave me valuable advice about the MCU I was planning to use. 
Many thanks and I will continue my investigation.

akhodeir, 2023-10-17:
Issue completed.

closed 2023-10-17 by akhodeir