iNavFlight/inav #9383: Bootloader porting

akhodeir, 2023-10-21:
I started the steps to port INAV on Pi Pico. 

My first step is to work on the Bootloader. I understand that the bootloader shall communicate the sensor readings with INAV configurator App and shall download the configuration of the flash.

I would like to ask, where can I find the existing protocol to communicate between INAV and the target device ?

Scavanger, 2023-10-21:
There is no "INAV bootloader" and no corresponding protocol.
 
INAV runs directly on the hardware, to port INAV to another processor you have to port the complete drivers (see https://github.com/iNavFlight/inav/tree/master/lib/main), adapt all low level accesses to the hardware (also after *STM32F* files in the code and adapt the cmake files accordingly).

akhodeir, 2023-10-21:
I didn't get it.

How the INAV configuration App talk to the device while configuring the sensors and the receivers??

Scavanger, 2023-10-21:
MSP (Multiwii Serial Protocoll):
However, there is no fixed documentation on this, see here:
https://github.com/iNavFlight/inav/tree/master/src/main/msp 
and 
https://github.com/iNavFlight/inav/blob/master/src/main/fc/fc_msp.c

But what do you want with it? INAV is supposed to run on the Pi Pico, and so far it runs on SMT32F4/F7/H7 and At32F4 microprocessors. The Pico Pi on the RP2040 developed by Raspberry itself would be a lot of work to port INAV to it.

akhodeir, 2023-10-21:
I think we talk about two different thing !!

MSP is related to transfer sensors information to other devices via serial port.

If you watch this video for example : 
https://youtu.be/yzYSd1MHfz8?feature=shared&t=55
you will find that flight controller is providing the sensor information to the computer via USB. This shall not be MSP.

I am talking about the infrmation that is shared via USB to a computer device for the configuration.

what kind of protocol is it ?

stronnag, 2023-10-21:
Regardless of your ill-informed comments, it's still MSP.

akhodeir, 2023-10-21:
> Regardless of your ill-informed comments, it's still MSP.

I am not sure why you insult me even if I respect that you are supporting me !

Scavanger, 2023-10-21:
Also MSP. 

MSP can (!) also be used as a sensor protocol (sensor <-> flight controller) via UART (but usually the sensors are addressed directly, without MSP), but MSP is mainly used for configuration (FC <-> Configurator), which is also the original use, just have a look at the sources I linked above.

closed 2023-10-21 by stronnag

akhodeir, 2023-10-21:
RP2040 is very cheap (almost $1) and quite available even in the corona panadimic. It has two M0+ cores. 
It can be connected to 16Mbyte of flash memeory which makes no limitation to the program or features requred. 

I have intensive experience with it since I used it in many projects. I am not familiar with STM32 and I hate its limited flash memory and its price range.

I have free time for two months and wanted to start a project with RP2040. I thought to port INAV to it and build FC. The price of the FC will be cheaper and maybe I can sell some of my FCs.

Scavanger, 2023-10-21:
Ah OK,

but as I said, MSP has nothing to do with it, you have to adapt all hardware accesses and processor-specific things accordingly, starting with timers, clock, GPIO, PWN etc. up to the interfaces such as UART, I2C, SPI, I2C.

The last port was to the AT32F4, which is quite similar to the STM32FX, you can have a look at the pull request:
https://github.com/iNavFlight/inav/pull/8752

To understand the dimensions:
210 New files
145,029 lines of code added
238 lines removed

akhodeir, 2023-10-22:
Thx for explanation