iNavFlight/inav #9392: Overcomplicated PIDFF tuning in fixed wing

cptX, 2023-10-23:
Hi, I'm an electrical & computer engineer and I have studied PID control in my studies and implemented it in some applications. I also am an experienced pilot on fixed wings, helicopters and recently multicopters and FPV. I got a Nano Talon which according to my experience is one of the worst aircrafts I have flown so far, meaning that to fly nicely it should have a really good PID tuning. So I installed INAV and started learning it and I came to the point I wanted to make the PIDFF tuning. After studying the little infos available in the internet (which are contradictory in many cases due to the changes in inav autotune and the removal of D, addition of FF and readition of D) I tried to tune it in the air. Nothing worked according to my previous PID tuning experience. I understand that the delay induced by the servos can affect the behavior. My attempts gave really pure results.

So to cut the long story short after studying again as many infos I could find in the internet about tuning INAV and after reading threads and discussions and especially this comment from @Avegawanderer
https://github.com/iNavFlight/inav/issues/3732#issuecomment-667725745
I realized that the problem was not me but probably the complexity and the overparameterization of INAV control system.
By this I mean that there are too many free parameters that interact to each other and they cannot be determined clearly.
For example tuning should start by setting the roll, pitch and yaw rates. But I don't know them!! If I set them too high AUTOTUNE will put too much FF to reach them, and maybe never reach them. Also in flight I may don't want to make full loops and full rolls and I may don't want to pull the sticks to the extreme, because maybe my model doesn't allow it or is too unstable. So here is the first caveat.
The second problem I realized is that increasing the rate increases the surface throw. Increasing the P increases again the throw. Increasing the FF again increases the throw. So you have 3 parameters increasing the throw and these are way too many!
Also FF Autotune has no good meaning because it will auto-adjust the values according to the already existing rate and P values. So again FF will not be deterministic! It doesn't make sense at all! Actually I have a really hard time to understand the need for FF because I thought PID is enough. By the way I saw the D need in the air so it definitely is needed and again I don't understand why it was removed in some INAV versions... I'm glad it is back though. I can trust the more experience pilots considering the FF useful so let's say we keep it but we need to find logical percentages between FF and PID control because otherwise doesn't make sense! For example let's say we need FF to be 50% of the input and the rest to be autocorrection from PID.
Now things are really bad. I ran autotune and for example pitch FF was 210. Almost the full of the scale, this leaves too less room for the PID! I really don't get it. It's impossible to tune my aircraft.

So, not knowing the rates and being completely free to select whatever value for the FF I'm really not in position to tune the aircraft. All these confuse the unexperienced pilots and I'm not considering myself a newbee in these fields.

What will happen if we avoid FF completely (FF=0)? What if we fix it to 50%? What will happen if FF has the value of 255?
How bad does the aircraft fly without FF? How should I define the values of FF? I don't like the Autotune for the FFs because they will be based on the set rates and P values so if you want to modify the P values you cannot have them defined before defining the FFs. It's a never ending and pointless loop. Maybe it would be more clear to say I want 50% FF and 50% PID and then fly and try to modify the PID to achieve this!

Please give me your lights. My brain has been shortcircuited the last days from this complexity...

Avegawanderer, 2023-10-23:
Greetings for everyone!
@cptX , I also felt frustrated when I tried to setup INAV for the first time at FW, though I had some experience in regulating systems, too! I ended up looking into source code.
I think the first thing to discuss here what the rate actually is. Rate here is the aircraft's ability to make turns faster or lower, in other words measure of agility. Since at higher speeds aircraft can make turns faster than at lower, rates differ for different speeds. 1m wingspan flying wing at say 60-80kmh will have typical roll rate of 300 - 400 degree (one turn) per second, while at 120kmh it may have 600 or more degree per second (with the same aileron deflection). 2m glider will have about 60-80 degree per second at similar speeds.
INAV simplifies this fact to a single fixed rate which should be selected by pilot for a speed he prefers. When I start tuning new aircraft, I usually fly manual at medium speed with blackbox enabled, do some turns and loops and later observe what the rate was for each axis. This (averaged) value becomes the rate setting in INAV. After that I zero P, I, D and adjust FF to get **the same deflection** as in manual mode (making sure the still is some margin to increase deflection- this is the room for PID loop). FF in INAV is simply RC control with some gain which also depends on rate. Yes, the selected rate is fixed and you setup INAV for a single airspeed so at other airspeeds you cannot get ideal regulation a priory . Maybe one day somebody will introduce varying rates depending on airspeed sensor but this is for now.
As I wrote at [#3732](https://github.com/iNavFlight/inav/issues/3732#issuecomment-667725745), I do not have any glue for why this rate setting scales PID coefficients. If I want to control a slow glider which has rate of, say, 40-50, I have to set a huge FF, maybe even limited by 255. But for fast agile wing which has rate of 400 P values will be like 10-20.

cptX, 2023-10-23:
Hi @Avegawanderer, thanks for your immediate response!
What if I cannot measure the rotation rate? My airplane is very unstable and I'm afraid to do full rolls and loops. How can I determine the rates? What if I put some rates out of my head?

What happens if the ff value is 255? Is there any room for the PID or will it cancel it out?

Also you say "I zero P, I, D and adjust FF to get the same deflection as in manual mode". You do this on the bench or in the flight? Because deflection could be visible on the bench too, right? Also if you put so much FF in order to have the same deflection as the manual, then how much room it remains for the PID to play it's game? Is too much FF cancelling PID or not? If I put too high values of FF and then the PID has less effect on the aircraft what will be it's capability in angle mode? Will it have enough deflection to fly correctly in autonavigation modes?

Avegawanderer, 2023-10-24:
"What if I cannot measure the rotation rate? How can I determine the rates? What if I put some rates out of my head?" - Well, you actually do not have to do full roll or pitch turn. It may be sufficient to do only a part of this maneuver as long as the airplane reaches it's rotation speed. For example, you fly -30 degree banking to the left and then move the stick to full right, the airplane turns to the right banking. In logs you would see as roll rotation increases from near 0 (if you were stable flying -30 banking) to some constant value and remains (of course there would be some noise) until you return the stick to center (thus stopping rotation). Doing a full roll (or maybe a few rolls) just provides more time to analyze the rotation speed. The goal is to make sure the turn lasts long enough for plane to achieve a constant rotation speed. Is is OK to start with rates out of your head. As any control loop, PID in INAV will output zero when the setpoint and measured value are equal (skipping I term for now). So if you have the FF component such that ail deflection is similar to manual mode, you can fly, do some partial rolls as described and later observe PID state comparing requested rate and actual rate. If PID is tuned fine, the rates will be quite close and PID will have near zero output. Doing the same turn while flying slower will cause the actual rotation rate be lower than setpoint (set by stick) and PID will try to increase deflection in order to reach setpoint. Similarly, turning while flying faster, actual rotation rate will be greater than setpoint and PID will try to decrease deflection.
So if you do not know the rates and do not have even an estimation, start with default value, adjust FF, fly and check the logs.

"What happens if the ff value is 255?  Is too much FF cancelling PID or not?" AFAIK 255 is just a limit for the FF gain and PIDFF does not have this limitation internally. PIDFF will not saturate and will continue to output the sum of FF and other components. But what actually does limit is the servo limits, so if you set the FF gain such that servo throw is at it's limit (setup in output/servos section) then yes, PID has room to decrease throw only, not increase. There must be some room to also increase deflection (this is where the the 90% recommendation comes from - assuming that in manual mode servos are setup to absolute maximum deflection). Of course, the initial setup of FF is done on the bench, just switch back and forth between manual and acro and observe servos.

"Also, what would happen if we completely zeroed out the FF values and increased the P values to have the same deflection. Wouldn 't the flight be equaly responsive?" This is how it used to be before FF was introduced. Cannot state anything as I didn't give a try. I personally do no use autotune in favor of procedure described above.

cptX, 2023-10-25:
Thanks @Avegawanderer, your time and effort to help and to explain are highly appreciated!

You gave me a lot of ideas of how to measure the rates.  I'll try to be methodical and do the calculations.
I have a question though: in some references in the internet it says that autotune changes the FF values AND the Rate values. So if it changes the Rate values why do I have to measure them myself and set them?

Also I have previous suggestions of yours to decouple the Rates from the PID. Do you think this could be possible? Then how the system could know the setpoint of the rates?

b14ckyy, 2024-03-22:
Sorry for not reading everything, its just a bit too much to answer all at once. But I think your Studying of PID control Systems makes you overthink the INAV system a bit. It is much simpler than you think. 

First of all, the PID component in INAV only acts as a small stability assistance. Unlike on Quads where the whole flying capability relies purely on the PID loop, in INAV the PID controller barely matters on fixed wings. 

Its the maximum rates, the commanded rates that scale from 0-max rates and the Feed Forward that do 90% of the work. 

FF is a passthrough component. There is no sensor data involved for FF. Simplified: 

Stick Input > results in Setpoint rate x Feed Forward = Mixer output. That simple. 

So feed forward in combination with the setpoint rate does the main control surface deflection. the PID loop then just assists and compares the setpoint with the gyro data end helps to maintain the target rate. 

In any autolevel mode the only difference is that the setpoint then comes not from the stick but from the attitude controller (purely proportional). 

So all you need to do is an autotune. Autotune automatically balances the max axial rate and feed forward, so that feed forward uses about 80% of the mixer range (by default) and measures the gyro rate achieved to store it as the maximum rate. After that the main tuning is done. 

then the PID tuning can be used to schange the fligh feel. Make the plane more agressive to fight gusts and turbulence or make  sharper or softer stick responses. Be more or less locked in when flying in acro and releasing the sticks. Compared to other systems INAV is really simplified. In most builds you can completely ignore the PIDs and just leave them on default. 

A few caviats are there but that's a design issue with some planes. Especially V-Tails have very non-linear pitch curves. sensitive for small motions of the control surface but then less sensitive on more agressive ones. Often because the high up Tail control surfaces stall in the turbulent wake of the wings. This can mess up autotune a lot and overtune feed forward with low measured max rates. Overtuned FF then results in massive oscillations and woblly flight behavior in self leveling modes. in this case you can manually tune that and I have a video for that here: https://youtu.be/VhIZ1F1UCEU

closed 2024-03-22 by DzikuVx