iNavFlight/inav #9396: New Flight Mode Soaring Drift INAV

dayyancom, 2023-10-23:
## Current Behavior
I was flying Glider in 2 condition calm air and so much windy, the result is SOARING MODE on INAV right now is only suitable for calm condition, but cant in super windy. Because it always maintain the position hold based on the GPS and will not have thermal drift when air is flowing into core thermal. When it pushed by the wind, glider will go so fast, but when making turn to maintain position based on GPS it will superr slow move because it have to penetrate the air and it can stall if in windy condition. (Look At Poor and Stall Glider Video Below)
[flightlog2kml-1.0.8-win64.zip](https://github.com/iNavFlight/inav/files/13070131/flightlog2kml-1.0.8-win64.zip)

## Desired Behavior
Soaring (Thermaling) Glider should be always following where the thermal goes, not fighting the thermal to maintaining the position, it will cause stall. Yes it still getting uplift but only in calm wind condition, and gonna be super picky in uplift thermal (only strong upwind can make glider goes up (look at Good Lift Video Below)

## Suggested Solution
Why not building a new Flighht Modes Just like HORIZON MODE where it stabilize 0 roll and 0 pitch to make plane fly straight, but in here what glider user want is holding roll angle (bank angle) and pitch angle, say it will hold 35 degree roll bank angle and 5 degree pitch down.

it will have advantages than the old SOARING MODE (POSITION HOLD + SOARING PITCH DEAD BAND) its will likely to be impossible to having stall when loitering (thermaling), it will have soaring drift as the air current flow into core thermal, and it will have better uplift triggering for gliders.

## Who does this impact? Who is this for?
Fixed Wings (Glider User)

## Additional context
Good Lift in calm wind = https://www.youtube.com/watch?v=o6j_MkQfRsQ&t=568s
Poor and stall super windy = https://www.youtube.com/watch?v=Ih9qtk5hLX4&t=110s

[flightlog2kml-1.0.8-win64.zip](https://github.com/iNavFlight/inav/files/13070151/flightlog2kml-1.0.8-win64.zip)

breadoven, 2023-10-24:
Yes the PosHold aspect of Soaring mode isn't so useful if it's windy or in general I guess if you're thermalling since thermals rarely hang around in the same spot. Ideally you'd want something like Ardupilot thermally mode which seems to work but that's probably a lot more difficult to do.

https://github.com/iNavFlight/inav/pull/9294 might be more useful for this although not tried it to find out.

b14ckyy, 2024-03-22:
Does Attitude hold fulfill your needs? I will close for now as there was no feedback for a while, @ me if you think it does not work.

closed 2024-03-22 by b14ckyy