iNavFlight/inav #9397: Enable Compass calibration via Inav programming/flight mode

sdellava, 2023-10-23:
## Current Behavior
Compass calibration can be done only using the configurator or via OSD Menù Stick.

## Desired Behavior
Enable to call the calibration via dedicated programming function.

## Suggested Solution
Compass calibration under certain conditions can produce random results. This can happen when the drone is close to large metal masses. 
In a previous feature request, I asked for the compass calibration procedure to be carried out during flight, so that we are certainly far away from metal masses and can calibrate the compass considering the weak magnetic fields generated by the drone's electronics. However, there was no activity and I think this stems from the fact that it is difficult to define a valid procedure for any type of drone, aircraft, etc. 

However, the solution is very simple: allow any pilot to activate the compass calibration procedure with a fligth mode or a programming task.

The procedure is exactly the same as that performed manually on the ground with the drone unarmed, so nothing new needs to be developed. You only need to create a task or fight mode that will be associated with an AUX, with which you activate the calibration procedure, simulating the pressing of the appropriate button on the configurator.

Obviously, to achieve a good calibration, the pilot will have to fly the aircraft by performing looping and tonneau in all axes. Each pilot will decide how to perform this manoeuvre in the allotted time, choosing an appropriate altitude. 

This solution should be simple to implement and leave the pilots free to decide whether to use it or not.
 
## Who does this impact? Who is this for?
All Inav pilots that use compass (RTH and Navigation)

sdellava, 2023-10-25:
added a PR: https://github.com/iNavFlight/inav/pull/9411

b14ckyy, 2024-03-22:
Nice one, closing as merged

closed 2024-03-22 by b14ckyy