iNavFlight/inav #9405: Minimum GPS air/3d speed for fixed wing navigation

machatron, 2023-10-24:
## Current Behavior
The current way navigation cruising speed is controlled is very blind because it isn't controlled at all. The autopilot is just blindly cruising at a preset throttle and linearly adjusting it depending on pitch. Throttle != speed. The same aircraft can fly fast with low throttle or slow with high throttle depending on environmental conditions such as headwinds and tailwinds.

## Desired Behavior
A config parameter to set a minimum speed that fixed wing navigation modes should fly in. If cruise throttle is set to 50% for example, but GPS air speed is below the configured threshold, cruising throttle will be increased until the minimum speed is met.

## Suggested Solution
A config parameter in cm/s units with the value being the minimum speed. In the configurator this could be in the "Fixed Wing Navigation Settings". The implementation could work well with a simple PID controller. This setting would be a simple "boost" on top of the cruising throttle similar to how pitch-to-throttle-ratio applies over cruising throttle. Obviously cruising throttle will not exceed max throttle even if the minimum speed could not be met.

## Who does this impact? Who is this for?
Me and anyone else who may be flying in poor, unpredictable, or altering weather conditions. This issue extends from #8918. Although I understand better the difference between relative ground and air speed and that this is not a perfect "anti stall" measure, I still disagree with much of what is said. I still think this is a good feature to add and is something I want on my fixed wing models. This is useful for people who want to balance flight time and distance covered. I like flying long distances but in headwinds I don't like waiting around forever to reach my destination because of winds, and enforcing minimum speed could assist with that.

## Additional context
This is a MINIMUM speed enforcement, not a cruising speed enforcement. This would not apply any additional throttle if the aircraft is cruising faster than the set minimum speed threshold. However I do believe that a long term goal for a controller to enforce an overall speed wouldn't hurt. Currently cruising speed is not controlled at all and I think that such a blind autopilot is not a good design. More customizability and control for how the autopilot fly's by leveraging the data from the sensors to make a craft fly more consistent despite environmental conditions is better. What if the autopilot could reach waypoints on a time schedule and not "whenever" the cruise throttle decides it'll get it there. I'm not saying I want this feature (yet) but controllers for things like travel speeds are the building blocks that can work up to such capabilities. These are the things that I like from more advanced autopilots like PX4 and ardupilot.

MrD-RC, 2023-10-25:
There is no such thing as GPS air speed. The GPS calculates ground speed. INAV already has a minimum ground speed. The only thing I would take from this issue is to maybe make that parameter editable. Though it maybe shouldn’t even be in configurator. As it can cause problems if set incorrectly. 

Also, the wind has no effect on the aircraft, once in the air. You say you disagree with what has been said in #8918 but they are the facts. Minimum ground speed has nothing to do with preventing stalls. Only your cruise throttle can help with that. All the minimum ground speed will do is make sure that you can fly in to wind and cover ground at a minimum speed. Nothing more. If you are stalling, your cruise throttle is too low. You really need to understand the difference about how the aircraft sees the air and how you see it.

breadoven, 2023-10-25:
This is just an adjustable minimum ground speed rather than the fixed value currently used to increase throttle when ground speed drops too low, currently set at 7 m/s.

Machatron, 2023-10-25:
> There is no such thing as GPS air speed

Slight terminology mix-up. You know what I meant.
> You say you disagree with what has been said

Reread what I said more carefully. I disagree with *much* of what is said not all. I explicitly stated I better understand my fault in the stall characteristics I mentioned. Nowhere here did I call this feature an "anti-stall" measure. The part's I disagree with is moving dynamic environmental configuration responsibility onto the user rather than the aircraft having flight characteristics adapt to external conditions without the need for profiles or configuring on the ground. 
I've stated here in detail *why* I want this and think it's a good idea. I don't like hardcoding values (like the 7/ms) when those have to do with flight physics and are arbitrarily chosen to be "universal" for most fixed wing. Some weirder models may want this tuned. And exposing such tunability and control opens the gates for flexibility in future expansion of the autopilot capabilities.

About the current 7 m/s, I agree exposing this may lead to issues. It's good the autopilot has *some* lower bound but I'm not sure if this is PID controlled and if this hard limit exists for specific edge case scenarios then maybe this limit should be it's own thing entirely and not just exposing the existing 7m/s value.

MrD-RC, 2023-10-25:
I never said anything about minimum ground speed stopping stalls at all. Ground speed has nothing to do with stalls whatsoever. All the minimum ground speed will do is ensure you will make some progress in to a headwind.

For example, your cruise throttle equates to a 40kmh airspeed. You fly in to a 30kmh headwind. Your ground speed would only be 10kmh. The minimum ground speed of 7m/s (approx 25kmh) will add throttle so that the ground speed is 25kmh. This would equate to a 65kmh airspeed.

PR #9428 (2023-12-06): Make min ground speed setable a parameter
This PR changes the hard coded minimum ground speed of 7m/s to a parameter. The parameter is in m/s and can be set in the ~~battery profile~~ navigation settings. Default is the current 7 m/s.

Fixes #9405