iNavFlight/inav #9415: Skystars F405HD2 number of motors output not correct - there are 6 outputs and not just 4 (free S5 and S6 pad available on the FC to solder)

Kastekniv88, 2023-10-26:
Hello,

I own an F405 hd2 by skystars, iNav target SKYSTARS405, but in the outputs tab I can find only 4 motors output instead of 6. I know there are additional 2 output motors because of the S5 and S6 pad you can solder on the FC itself. I would like to use 1 of these motor pad to install a servo motor, otherwise I don’t know how to use it, I have also chosen this FC for this opportunity. Can you please let me know of there is the possibility to change the pwm output motor from 4 to 6? Thank you so much!

b14ckyy, 2024-03-22:
The pad assignments for S5 and S6 in Betaflight are 

resource MOTOR 5 B00
resource MOTOR 6 B01

With my bad luck editing targets I hesitate to touch this but @shellixyz made the original target back then, maybe he is around and can look into that?

MrD-RC, 2024-03-22:
Very unlikely. @shellixyz moved to ardupilot dev. Now is making his own firmware because ArduPilot is a piece of sh1t. 

Maybe I can take a look at some point.

MrD-RC, 2024-03-24:
> The pad assignments for S5 and S6 in Betaflight are
> 
> resource MOTOR 5 B00 resource MOTOR 6 B01
>
No idea what BetaFlight are doing in that case. There are no **B00** or **B01** resources. There is **PB0** and **PB1**. But nothing beginning **B0** or **BO**.

MrD-RC, 2024-03-24:
For future reference. The BetaFlight unified target fucks with the resource IDs. They remove the initial **P** and add a leading zero to the number. So a resource ID of **PB7** becomes **B07**. Why!? 🤦🏻

MrD-RC, 2024-03-24:
@Kastekniv88 I don't have this flight controller. Would you be able to test the outputs? Of course, make sure there are no props on when you try this.

[inav_7.1.0_SKYSTARSF405HD2.hex.txt](https://github.com/iNavFlight/inav/files/14736358/inav_7.1.0_SKYSTARSF405HD2.hex.txt)
Just rename the above file to remove the `.txt` at the end.

Yarolsav180703, 2024-09-18:
> @Kastekniv88 I don't have this flight controller. Would you be able to test the outputs? Of course, make sure there are no props on when you try this. [...]

I just tested it on my SkystarsF405HD4 and it all works well with motors
Mapping:
S1 Timer4
S2 Timer4
S3 Timer3
S4 Timer2
S5 Timer3
S6 Timer3
It all can work well for motors, but if you want to add servos, you need to redefine S5 and S6 to another timer or resolder outs

Can I ask you to compile that HEX with another timer on S5 S6 for check, please?

closed 2024-10-30 by DzikuVx