iNavFlight/inav #9417: high throttle oscilations speedybee f7v3

11BEX11, 2023-10-27:
hello, i have speedybee f7v3 in my speedybee master HD frame...i just cant get the high throttle oscilations to stop. Even on quick punch it sounds strange...I have 6s  Diavola 2207 / 1960 kv motors with 5inch T-motor T5147 props installed...I changed props (high/low) pitch props..gemfan and T-motor.. I used pid toolbox for pidtuning (basement tunning) and it was realy good tune on " paper "  but still on high throttle it flutters..Bad propwash also...is there anyone to FLY me in the right direction..I hope i am on the right place. is that just bad filtering or is there some issue with speedybee f7v3 stack. and let me point out that in betaflight there was no issue just good flying...i realy wanna use inav because i would like to have opflow lidar and poshold and those things..its just better for my taste.. if you need more info tell me.. thanks if anyone answers to me.. have a nice day and happy flying

Jetrell, 2023-10-27:
Have you tried increasing the `TPA_RATE` to attenuate the gains at higher throttle levels.

11BEX11, 2023-10-28:
hmmm i will give that a try..thanks for the tip

closed 2023-11-09 by DzikuVx