iNavFlight/inav #9418: Non disarm on Landing RTH

Alexh2807, 2023-10-27:
****

## Current behavior
When the drone performs an automatic landing, with RTH for example. The drone hits the ground, it takes time to disarm, and bounces back, while transmitting to corigger its GPS positioning at the same time so it rubs against the ground, I tried to increase the descent speed up to 2m /s, to set the "bumb detection" with the CLI but it was already activated. and I also increased the nav_land_detect_sensivity" from 5 to 8, but the same observation. the drone shuts down after a few seconds after touching the ground, if it does not move too much with the GPS positioning which remains constantly activated even after touching floor.

(I use Google translate)

## Steps to reproduce

1.Activate RTH with a SpeedyBeeF405 v3
2.Wait for the moment when the drone lands
3. possibly notice rebounds

## Expected behaviour
That the drone cuts off the engines when it receives an impact greater than a chosen value during the landing phase

## Suggested solution(s)
I think this is already the case but hey, having an option which allows the engines to be cut off as soon as an impact is felt during the landing phase, as well as security which deactivates GPS positioning when it is not not disabled. the motors after an impact as well, in case the drone continues to rotate while sliding on the ground

---

- Name and supplier of the FC Board: SpeedyBeeF405 V3
- INAV version string: 6.1.1

Jetrell, 2023-10-27:
It takes a couple of seconds for the landing detector to measure that the multicopter has landed.. What you are seeing is normal behavior.
But you can lower the `throttle_idle` value to prevent it skidding across the ground, before it disarms. When Air mode is permanently enabled.
The 7.0 release will have a multicopter landing bump detector. Which will instantly disarm your quad at touchdown.

breadoven, 2023-10-28:
Bump detection might help but sometimes ground effect slows the descent enough in the last 20cms or so to give a touchdown so light it doesn't trigger the bump detection. Normally rises back up again and tries again with more of a bump which usually works second attempt. But no bump detection in 6.1.1. so it was just relying on the average throttle during descent method which seems slow to react especially if ground effect causes bouncing. Is there some setting to prevent this @Jetrell, some MR's don't seem to do it.

Jetrell, 2023-10-28:
> Is there some setting to prevent this. some MR's don't seem to do it.

There are some things that can be done to mitigate it. Like ensuring there is enough slowdown time between `nav_land_slowdown_maxalt` and `nav_land_slowdown_minalt`.. And setting `nav_land_minalt_vspd` accordingly.
 And not having `mc_p_level` set too tight.

I've also found the Integral value on the pitch and roll axis's influence stability once it touches down, and waits for a landing to be detected.  

But it would seem the biggest factor is the motor torque and combine propeller disc area, verse the weight of the multicopter..

Alexh2807, 2023-10-28:
For the moment I have lowered the value of Trottle Idle from 9 to 8.5% (at 7% the motors stop).
I changed the `nav_land_slowdown_maxalt` from 2000 to 2500.
The `nav_land_slowdown_minalt` from 500 to 1000,
and the `nav_land_minalt_vspd` which I had set to 200 to see if even with a strong enough shock it would stop, so I lowered it from 200 to 50.

However I don't understand the `mc_p_level` too tight, should I rather increase it at least? And what would be its influence (In any case I have value and at 20)? And for the Integral value?

If I can provide a little more information, my drone is a 5 inch, 600 to 650 grams, with 5.1 inch pitch 3.5 propellers.

In any case, thank you for your help!

Jetrell, 2023-10-29:
> 
> However I don't understand the `mc_p_level` too tight, should I rather increase it at least? And what would be its influence (In any case I have value and at 20)? 

20 is a good starting value. Often having it set too high. Mid twenties and upward. Can cause problems.

> And for the Integral value? If I can provide a little more information.

This will depend on the physical properties of the quad..  Higher integral can cause windup in AIR mode, when the Quad is on the ground, which isn't desirable.. But it can also aid the (Nav) ANGLE mode stabilization response for quads that have an under-slung battery..

Remembering, that landing in windy conditions will always cause problems too... When a wind is blowing horizontally over the propellers. This will increase their lift. Which in-turn will make the quad lighter at touchdown, and make it more difficult for the landing detector to sense that it isn't still flying.

closed 2024-03-22 by DzikuVx