iNavFlight/inav #9439: Current Behavior
Hi MAMBA F722IC ports tab on Inav 7 rc1 do not let to make the same changes as it was on inav 6. First photo is status from inav7. Second photo is what i want to set on inav 7. Third photo what was set on inav 6 and it was working. When i try to put frsky with serial rx and then save and reboot it just reboots to default.
Gps i am using is geprc m10 dq.

xnerkax, 2023-11-03:
[image]

[image]

[image]

MrD-RC, 2023-11-03:
You shouldn’t be able to set telemetry and Serial RX on the same UART. What is your rc link setup?

xnerkax, 2023-11-03:
Im using teranis qx7 with r9m with fport

MrD-RC, 2023-11-03:
Then you just set SerialRX and in the receiver page set the protocol to FPORT. You don’t set anything on the ports page in the telemetry column.

closed 2023-11-03 by MrD-RC