iNavFlight/inav #9440: UBLOX M6N not working after upgrade to INAV 7.0.0-RC1

ToughAce, 2023-11-03:
## Current Behavior
UBLOX M6N not working after upgrade to INAV 7.0.0-RC1 on MATEKF405TE
Hardware health check fails

## Steps to Reproduce
1.  Flash INAV 7.0.0-RC1 to MATEKF405TE
2. Enable GPS on UART5
[image]
3. Enable GPS for navigation and telemetry feature
[image]
4. CLI command: set gps_ublox_nav_hz = 5
[image]
5. GPS Hardware health check fails

## Expected behavior
Same GPS works fine with same FC INAV 6.1.1. Expect it to work in INAV 7.
[image]

## Suggested solution(s)
Please review and fix GPS implementation

## Additional context

https://pastebin.com/XmKaBxGK

---

- FC Board name and vendor:
- INAV version string:
# version
# INAV/MATEKF405TE 7.0.0 Oct 31 2023 / 13:19:54 (fdb61d3a)
# GCC-10.3.1 20210824 (release)

stronnag, 2023-11-04:
Confirmed. 

@mmosca is this intentional (in which case we should announce it in the RN), or an unforeseen side effect of the other GPS changes?

* Happens regardless of auto-baud / auto-config
* If I run mwp's `ublox-cli` in INAV `gpspassthrough`,  with `ublox-cli`'s `noinit` mode then nothing is seen
* If I run mwp's `ublox-cli` in `gpspassthrough`   mode (which applies config like INAV pre 7.0), then `POSLLH` is observed as expected.

So perhaps something in the new setup INAV applies upsets M6?

ToughAce, 2023-11-04:
Thanks @stronnag. How did you configure the M8N? I couldn't get my one to work either.

stronnag, 2023-11-04:
> Thanks @stronnag. How did you configure the M8N? I couldn't get my one to work either.

Absolutely nothing special. It's a really old model, UBLOX v2 firmware only (so circa 2016 I'd guess).

Non-default settings:
```
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set gps_ublox_use_glonass = ON
```
Otherwise defaults, auto baud / config.  

Thusly:

```
# status
INAV/WINGFC 7.0.0 Oct 31 2023 / 19:02:46 (fdb61d3a)
GCC-13.2.0
System Uptime: 18 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 10.64V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU9250, ACC=MPU9250, BARO=BMP280, PITOT=VIRTUAL
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x10010000, Heap available: 1828
I2C Errors: 0, config size: 9220, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 3, used = ADC 3
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 1, used = none
System load: 5, cycle time: 1010, PID rate: 990, RX rate: 9, System rate: 9
Arming disabled flags: NAV RX CLI
OSD: not enabled
VTX: not detected
GPS: HW Version: UBLOX8 Proto: 15.00 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    BeiDou 0/0
    Glonass 1/1
    Max concurrent: 2
```
`UBLOX8 Proto: 15.00 ` is pretty ancient.

stronnag, 2023-11-04:
^ That is the canonical bad actor, all my newer M8Ns with UBLOX v3 firmware also do Galileo.

mmosca, 2023-11-04:
I need to check. Maybe it is failing if we try to enable / disable galileo, similar to when we try to autobaud with "unknown" baud rates.

Does it work in UBLOX mode? @stronnag do you have hardware to reproduce this? Unfortunatelly, I only have newer units, so haven't run into this in my testing.

stronnag, 2023-11-04:
> I need to check. Maybe it is failing if we try to enable / disable galileo, similar to when we try to autobaud with "unknown" baud rates.
> 
> Does it work in UBLOX mode? @stronnag do you have hardware to reproduce this? Unfortunatelly, I only have newer units, so haven't run into this in my testing.

I have a M6 GPS (I've been doing INAV for (too) long as to have a substantial WEEE box of ancient stuff).
Outside of INAV's configuration, the device works fine with UBLOX.
I will dig deeper over the next few days.

mmosca, 2023-11-04:
A quick glace at the code, we at least check if it has the galileo capabilities before trying to set it up.
I did find a || true on an if statement that probably should not have been there.

Can you try commenting out lines 420 to 430 on gpx_ublox.c just to confirm if it is related to the gnss config?

You can do one at a time, or all in one go.

stronnag, 2023-11-04:
I flashed #9445, now in `status`. Max concurrent looks a bit strange. 
```
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10010000, Heap available: 1828
I2C Errors: 0, config size: 9220, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 3, used = ADC 3
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 1, used = none
System load: 5, cycle time: 1015, PID rate: 985, RX rate: 9, System rate: 9
Arming disabled flags: NAV RX CLI
OSD: not enabled
VTX: not detected
GPS: HW Version: UBLOX6 Proto: 0.00 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Max concurrent: 0

```
`GPS=None` but identified as M6 ???

mmosca, 2023-11-04:
I suspect this is
 in line with inav 6 behavior.
Do you have boot log from this gps?

stronnag, 2023-11-04:
boot log ?

mmosca, 2023-11-04:
I think you may be able to get it with u-center. The bootlog is the message it shows when booting that has the software version and capabilities.

stronnag, 2023-11-04:
> I think you may be able to get it with u-center. The bootlog is the message it shows when booting that has the software version and capabilities.

OK, I'll take a look.

stronnag, 2023-11-04:
There are no extensions listed, the SW and HW versions are:

SW: [7.03 (45969)] 
HW: 00040007 40007

Consistent with UBLOX6 in `status`.

mmosca, 2023-11-04:
I pushed a new change on the branch, can you check if it works with m6n gps?

mmosca, 2023-11-04:
@ToughAce can you try this firmware?

[inav_7.0.0_MATEKF405TE_ci-20231104-becec53.hex.zip](https://github.com/iNavFlight/inav/files/13257748/inav_7.0.0_MATEKF405TE_ci-20231104-becec53.hex.zip)

ToughAce, 2023-11-05:
Thanks @mmosca! Unfortunately still doesn't work:
[image]

I can dig into gpspassthrough and u-center later to provide some more details. Is this description still accurate to get it running https://github.com/iNavFlight/inav/issues/1332 ?

Phil-MC, 2023-11-05:
Surely the battery should be connected for the GPS to be visible in the CLI status report.

ToughAce, 2023-11-05:
I have GPS connected through the 4,5V rail that receives Power through USB. See also INAV6 status.

stronnag, 2023-11-05:
Yeah!!!! NeoM6 and **working**
[image]

PR after I've started the bread making ... priorities ... priorities

Phil-MC, 2023-11-05:
OK

Phil-MC, 2023-11-05:
Have you confirmed power at the GPS with a meter?

ToughAce, 2023-11-05:
> Have you confirmed power at the GPS with a meter?

Yes. Also GPS LED lights up and it works fine as it is connected right now in INAV6.

mmosca, 2023-11-05:
> Yeah!!!! NeoM6 and **working** [image]
> 
> PR after I've started the bread making ... priorities ... priorities

Does it work woth ublox but not ublox7?
What was the issue?

stronnag, 2023-11-05:
#9446 
The 6M never got passed the get capabilities phase, so PR skips it for 6M and earlier. 
Maybe there is a better solution, but this works and does not affect newer models.

mmosca, 2023-11-05:
Do m7 gps exist?Did it work for your early m8?This will likely prevent any GNSS configuration though. I will have a look at the documentation to see if we can filter by sw version.

#9446

stronnag, 2023-11-05:
I may have a M7N; they were utter crap (seems like avoiding odd numbered UBOX is a good plan).
Yes, as stated in the PR, it works fine with the ancient M8N.

stronnag, 2023-11-05:
And from the documentation, `MON-GNSS` is only supported on M8 and above, so I should amend the PR so it excludes M7N junk as well.

stronnag, 2023-11-05:
Also, from the UBLOX documentation, `CFG-GNSS` is onlu supported on M8N and later, so the configuring additional GNSS is moot.

ToughAce, 2023-11-05:
Did some more testing. My M8N works now with the fixed firmware:
```
# status
INAV/MATEKF405TE 7.0.0 Nov  4 2023 / 21:05:26 (dcbdf070)
GCC-10.3.1 20210824 (release)
System Uptime: 32 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06, PITOT=VIRTUAL
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=UNAVAILABLE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x10010000, Heap available: 1920
I2C Errors: 0, config size: 9551, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 5, cycle time: 1001, PID rate: 999, RX rate: 9, System rate: 9
Arming disabled flags: CAL ACC HWFAIL RX CLI
OSD: MAX7456 [30 x 16]
VTX: not detected
GPS: HW Version: UBLOX8 Proto: 18.00 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/0
    BeiDou 0/0
    Glonass 0/1
    Max concurrent: 3
```

Unfortunately still no success with my M6N. It's probably even more ancient than anything @stronnag has.
This is the version info from u-center connected through an FTDI adapter (couldn't get it to work with `gpspassthrough`):
[image]

stronnag, 2023-11-05:
That's because it's a M7N and you'll need the last commit to #9446 to fix that.

stronnag, 2023-11-05:
It is **much** newer than my ancient M6 @ToughAce

stronnag, 2023-11-05:
@ToughAce  There is nothing in #9446 that affects M8N and later, so that your M8N now works is just one of those things...

ToughAce, 2023-11-05:
> @ToughAce That's because it's a M7N and you'll need the last commit to #9446 to fix that.

The sticker on it says NEO-6M-0-001:
[image]

I tried this fix:
> @ToughAce can you try this firmware?
> 
> [inav_7.0.0_MATEKF405TE_ci-20231104-becec53.hex.zip](https://github.com/iNavFlight/inav/files/13257748/inav_7.0.0_MATEKF405TE_ci-20231104-becec53.hex.zip)

Where do I find the hex for #9446?

DzikuVx, 2023-11-05:
OK, Neo6M will be fixed in RC2

mmosca, 2023-11-05:
> The sticker on it says NEO-6M-0-001: [image]
>
> I tried this fix:
>
>
> Where do I find the hex for #9446?

Check for the artifacts here:
https://github.com/iNavFlight/inav/actions/runs/6760714159#artifacts

There should be a big zip file with the hex you need.

ToughAce, 2023-11-05:
Works for me now as well.
THANK YOU

```
# status
INAV/MATEKF405TE 7.0.0 Nov  5 2023 / 10:29:56 (31ff20aa)
GCC-10.3.1 20210824 (release)
System Uptime: 22 seconds
Current Time: 2023-11-05T11:10:14.884+00:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06, MAG=HMC5883, PITOT=VIRTUAL
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x10010000, Heap available: 1920
I2C Errors: 0, config size: 9551, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 5, cycle time: 1006, PID rate: 994, RX rate: 9, System rate: 9
Arming disabled flags: CAL COMPASS ACC RX CLI
OSD: MAX7456 [30 x 16]
VTX: not detected
GPS: HW Version: UBLOX7 Proto: 0.00 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Max concurrent: 0
```

mmosca, 2023-11-05:
FIx merged and will be included in rc2

closed 2023-11-05 by mmosca