iNavFlight/inav #9442: INAV 7.0.0 RC1 KAKUTE F7 motors output bug

capblack, 2023-11-04:
INAV 7.0.0 RC1 KAKUTE F7 motors output bug , motor 1 is not working and the 2,3,4  It's messy
quad x profile

mmosca, 2023-11-05:
Can you post a picture of your mixer tab? Did you check if the outputs match what you expect, like it is described in the release notes?

capblack, 2023-11-05:
[image]

Hello, this is what I have after loading the diff file from inav 6.1.1
I hope I'm not omitting something due to ignorance of the new mixer

mmosca, 2023-11-06:
That looks correct. Can you double check it still works in inav 6.1.1 and it is not a hardware issue. 

The motor allocation matches inav 6 and older, so I wouldn’t expect a change in behavior.

capblack, 2023-11-06:
Hi yes is correct on INAV 6.1.1

mmosca, 2023-11-06:
Turns out the new motor assignment code is too smart and exposes an old timer conflict
Thanks for reporting

capblack, 2023-11-07:
I have tried it as props and with this firmware it works fine. , then I will try it in flight

commit 9783a7b1 (2023-11-06): Remove PPM entry with conflicting timer.

Fixes #9442
closed 2023-11-07 by DzikuVx