iNavFlight/inav #9450: On inav 7 rc1 got mag qmc5883l issue

xnerkax, 2023-11-06:
Hi i have geprc m10 dq unit and have a issue with magnetometer that is shown in the picture, i dont have readings in every direction as you can see it dosent go below 0 but in inav 6.1.1 it does. Maybe someone has the same problem and knows the fix. Tried to calibrate numerous of time. The board is mamba f722 i2c[image]

shota3527, 2023-11-07:
Hi, this commit will fix this 8e7cc6bb32c34950d8074c1b47805d986978cd65

xnerkax, 2023-11-07:
Hi i made the changes now it shows. If someone needs it is whit added dps310 baro and updated fc msp file.

xnerkax, 2023-11-07:
[mambaf722_i2c.zip](https://github.com/iNavFlight/inav/files/13283479/mambaf722_i2c.zip)

mmosca, 2023-11-14:
Have you tried with INAV 7.0 rc2?

xnerkax, 2023-11-14:
Yes i tried it the day it was released and it was the same

mmosca, 2023-12-01:
> > Hi, this commit will fix this [8e7cc6b](https://github.com/iNavFlight/inav/commit/8e7cc6bb32c34950d8074c1b47805d986978cd65)
> 
> Is this suppose to be included in v7 ? Or for anyone to selfcompile if needed?

Try rc3.

Thidsa, 2023-12-01:
Just what i wanted to know that its included allready. thx

closed 2023-12-05 by mmosca