iNavFlight/inav #9455: Mamba H743 V2 (4th Gen) User1 VTX power switch not working.

marcdornan, 2023-11-07:
User1 power switch function for the VTX is not working on Mamba H743 V2 (4th Gen) on iNav 6.1.1 using MAMBAH743_2022B target, It seems to be non functional as Pairan reported on March 2023. I tested it by flashing Betaflight and it worked fine: 9V is applied to the VTX power pin when User 1 is set up on the modes tab and a channel is mapped.

I found this issue opened by Pairan in March 2023 closed but is the exact one I have with the same target but using 6.1.1 rather than 6.0.
INAV6 RC2 : USER1 without function on MAMBAH743_2022B #8844
https://github.com/iNavFlight/inav/issues/8844

mmosca, 2023-12-05:
Nice to see it works for you now.
I will add some documentation for this target.

closed 2023-12-05 by mmosca

Snowhack, 2024-01-25:
What is the reason to do this in the first place.  I have this board with a Vista, I don't have anything set up with user 1 or 2.  I also do not have an image on my goggles, is that why? I was referred to this page by Diatone when I inquired with them about my issue.

mmosca, 2024-01-25:
> What is the reason to do this in the first place. I have this board with a Vista, I don't have anything set up with user 1 or 2. I also do not have an image on my goggles, is that why? I was referred to this page by Diatone when I inquired with them about my issue.

So you can turn the vtx off, either on the bench, waiting for gps fix, or for some other reason and prevent it overheating.

Snowhack, 2024-01-25:
So it's on then as default if I didn't set this up? Trying to track down why my Vista is not working on this FC when it did on the one I swapped it out for.

PR #9539 (2023-12-05): Add VTX power info for MAMBAH743_2022B target
Fixes #9455