iNavFlight/inav #9457: MAMBA H743 gyro and accel not recognized in inav 5, 6, 7, anymore after flashing multiple times trying to make osd of the o3 work.. if i flash betaflight gyro and accel are recognized and working

ghostdog3993, 2023-11-07:
## Current Behavior
MAMBA H743 gyro and accel not recognized in inav 5, 6, 7, anymore after flashing multiple times trying to make osd of the o3 work..
if i flash betaflight gyro and accel are recognized and working

## Additional context
Status:
INAV/MAMBAH743_2022B 7.0.0 Oct 31 2023 / 13:15:54 (fdb61d3a)
GCC-10.3.1 20210824 (release)
System Uptime: 24 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 1.71V (1S battery - CRITICAL)
CPU Clock=480MHz, BARO=SPL06, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 480 MHz
  HCLK   = 240 MHz
  PCLK1  = 120 MHz
  PCLK2  = 120 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=NONE, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x20020000, Heap available: 1944
I2C Errors: 50, config size: 9563, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = none, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 3, used = none
System load: 17, cycle time: 502, PID rate: 1992, RX rate: 9, System rate: 10
Arming disabled flags: ANGLE CAL COMPASS HWFAIL RX CLI DSHOTBEEPER
OSD: MAX7456 [30 x 16]
VTX: not detected

---
# version
# INAV/MAMBAH743_2022B 7.0.0 Oct 31 2023 / 13:15:54 (fdb61d3a)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 0 AUTO
timer_output_mode 1 AUTO
timer_output_mode 2 AUTO
timer_output_mode 3 AUTO
timer_output_mode 4 AUTO
timer_output_mode 5 AUTO
timer_output_mode 6 AUTO
timer_output_mode 7 MOTORS
timer_output_mode 8 AUTO
timer_output_mode 9 AUTO
timer_output_mode 10 AUTO
timer_output_mode 11 AUTO
timer_output_mode 12 AUTO
timer_output_mode 13 AUTO

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
[... 1995 lines cut ...]
set failsafe_throttle = 1000
set nav_mc_hover_thr = 1300
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# restore original profile selection
mixer_profile 1
profile 1
battery_profile 1
---
# version
# INAV/MAMBAH743_2022B 7.0.0 Oct 31 2023 / 13:15:54 (fdb61d3a)
# GCC-10.3.1 20210824 (release)

DzikuVx, 2023-11-07:
@ghostdog3993 please flash Betaflight and provide output of Betaflight `status` command, I'd like to confirm gyro they use over here right now

DzikuVx, 2023-11-08:
@ghostdog3993 could you please test this: 
[inav_7.0.0_MAMBAH743_2022B.zip](https://github.com/iNavFlight/inav/files/13295828/inav_7.0.0_MAMBAH743_2022B.zip)

Seems like Diatone randomly changes how gyro is conneced... Sometimes it's SPI1, sometimes SPI4

I have no hardware, I can't test

ghostdog3993, 2023-11-08:
i can for sure mate i may have time tonight, thank you and will let you know

ghostdog3993, 2023-11-08:
@DzikuVx hi Pavel i flashed the drone and the gyro and accel are now working, tomorrow i will do a maiden flight and test if everything works as intended but for now a HUGE THANK YOU.

closed 2023-11-09 by DzikuVx