iNavFlight/inav #9468: INAV 7.0.0 RC2  GPS set satellites  Galileo Beidou Glonass not take effect also from cli still off MAMBA MK4 H743 V2

capblack, 2023-11-09:
INAV 7.0.0 RC2  GPS set satellites  Galileo Beidou Glonass not take effect also from cli still off 
Board Diatone MAMBA MK4 H743 V2

MrD-RC, 2023-11-09:
The number and types of constellations you can set are defined by your GPS module. The FC doesn’t matter.

capblack, 2023-11-09:
Yes, I know that, just for reference I put the FC, the GPS is a Walksnail-GPS M10 with UBLOX7 protocol selected and the constellation selection does not work

mmosca, 2023-11-10:
You gps does not support all combinations.

If you select an invalid combination, inav will reset the constelations to the default settings.

It may have a maximum number of constelations supported at the same time (reported in status) and some combinations of constelatins may be incompatible with each other.

Ps.: If gps reports 2 maximum constelations, that means GPS + 1.

closed 2023-11-10 by mmosca