iNavFlight/inav #9473: add DSHOT_DMAR to the TMOTORF7 (not the v2)

Pi-wi74, 2023-11-10:
Inav7.0.0-RC1

Tmotor F7 board (not the V2)

When selecting Dshot 300/600, motor 2 don't start 

from mosca  / Discord server
there is a dma conflict

All is good with Multishoot

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: T-Motor F7 with STM32F722 RET6
- board is labelled "T-Motor F722"
- INAV version string: Inav 7.0.0 - RC1

DzikuVx, 2023-11-10:
PR is for proposing code changes, not reporting issues

closed 2023-11-10 by DzikuVx

Pi-wi74, 2023-11-10:
i'm sorry, i'm a newbie on Github ... what should I do to make Dshot 600 working for the T-Motor F7 ? 
It seems that there is a "DMA Conflict" as Moska told me on the Discord server ...