iNavFlight/inav #9474: INAV 7.0 RC2 - Speedybee F7V3 / Speedybee BL32 50A 4-in-1 ESC - ESC #4 not detected

chmod755a, 2023-11-10:
***

## Current Behavior
Installed on 7" quadcopter.  Everything was working on INAV 6.1.
After flashing INAV 7.0 RC2, quadcopter arms but motor #4 would not spin up.
Opened BLHeli Suite, connected and tried to read all ECUs.  Only ECUS #1-3 could be read.  ECU #4 timed out.
Flashed INAV 6.1 and quadcopter arms with all 4 motors spinning up. BLHeli Suite is able to read all 4 ECUs.

## Steps to Reproduce

1. Flash INAV 7.0 RC2 on Speedybee F7V3 FC with Speedybee BL32 50A 4-in-1 ESC connected
2. Arm quadcopter
3.
4.

## Expected behavior

Expect that FC is able to communicate with all 4 ECUs

## Suggested solution(s)

## Additional context

https://pastebin.com/hMdvtLVx
---

- FC Board name and vendor: Speedybee F7V3
- INAV version string: 
- INAV/SPEEDYBEEF7V3 7.0.0 Nov  8 2023 / 15:41:13  (a54849be)
- GCC-10.3.1 20210824 (release)

DzikuVx, 2023-11-10:
OK, so we have a second case of this

mmosca, 2023-11-10:
Pr #9475 changes the defaults to work for a quad, out of the box.

DzikuVx, 2023-11-12:
OK, will be fixed with INAV 7 RC3

commit 78ebfeb0 (2023-11-11): Change timer allocation default to MOTORS

This board has an unfortunate timer allocation, where the first 4 outputs use 3 timers.

This causes the default allocation to be M1, M2, M3, S1, M4

Fixes #9474
closed 2023-11-12 by DzikuVx