iNavFlight/inav #9476: Expanding view for the osd window

oldrootbeer, 2023-11-10:
Is it possible to have a 'hover over widget / button' to temporarily magnify the osd centre window? Like a pop up? Hover over toi magnify like Amazon images.

[image]

error414, 2023-11-13:
why do you want it? 

I was thinking about to keep preview fixed on page during scrolling.

oldrootbeer, 2023-11-13:
> why do you want it?
> 
> I was thinking about to keep preview fixed on page during scrolling.

I'd just like to see it bigger. In my face like in the goggles with out the distraction of the settings either side. I have a 27" monitor, but still seems small. I guess my eyes are steadily going

b14ckyy, 2024-03-22:
Moving to Configurator repo. Changes in the GUI are planned, we can put this on the consideration list.