iNavFlight/inav #9487: iNav target JHEMCUF405 needed

Tigerlie, 2023-11-13:
## Current Behavior
i want to use this flight controller with iNav because i need only servos for a FPV RC car

https://www.aliexpress.com/item/1005005789651798.html?spm=a2g0o.productlist.main.35.1dbehcMShcMSkf&algo_pvid=5658691b-edab-454b-8325-2b0c8133f2a6&algo_exp_id=5658691b-edab-454b-8325-2b0c8133f2a6-17&pdp_npi=4%40dis%21EUR%2122.11%2121.89%21%21%21168.37%21%21%40211b61bb16999140781176079e8b74%2112000034362258699%21sea%21DE%214407378106%21&curPageLogUid=eCB17A2berph

can't find .hex target on iNav

## Desired Behavior
finding the correct target to put iNav firmware on FC and use with RC Car with O3 Air Unit (need the FC for OSD and maybe GPS and also to turn on high power mode (arming) for the O3 Air Unit

## Suggested Solution
also a RC CAR mixer would be cool, with only servos. (the Brushed ESC is controlled with normal servo PWM)

## Who does this impact? Who is this for?
the entire FPV RC car community

## Additional context
specifically i want to set up with my O3 Air Unit and V2 Goggles (no WTF OS)

mmosca, 2023-11-14:
Please follow #9437

Tigerlie, 2023-11-14:
i see, it seems like the flight controller developer needs to do some work.

closed 2023-11-14 by Tigerlie