iNavFlight/inav #9488: 7.0.0-RC2 - Wrong motor mapping in SpeedyBee F7 V3 BL32

agoliveira, 2023-11-14:
## Current Behavior

The automatic mixing for a quad using the 7.0.0-RC2 in the SpeedyBee F7 V3 BL32 FC is mapping motor 4 to S6 instead of S4.
[image]
If the timer outputs are set to MOTORS, the mapping is correct
[image]
This does not happen with INAV 6.1.1

## Steps to Reproduce
Install INAV 7.0.0-RC2 or RC1 in the FC, set initial mixer to multirotor 7".

## Expected behavior
The motors should be mapped as per the second image (S1 -> motor 1, S2 -> motor 2, etc).

## Suggested solution(s)
Mapping should be the same as 6.1.1, this is a regression.

## Additional context
Dump: https://pastebin.com/xQWZVnct

- FC Board name and vendor: SpeedyBee F405 V4 BLS 55A 30x30 Stack
- INAV version string: # INAV/SPEEDYBEEF7V3 7.0.0 Nov  8 2023 / 15:41:13 (a54849be)
# GCC-10.3.1 20210824 (release)

mmosca, 2023-11-14:
This was fixed in #9475 and will be include in the next release (rc3 or 7.0 final, whichever comes first)

closed 2023-11-14 by mmosca

agoliveira, 2023-11-14:
Thanks. I did search for it but wasn't able to find it probably because of the way the FC was mispelled.