iNavFlight/inav #9490: how to use inav on speedy bee f405 v3

Diywithall, 2023-11-15:
I am using a speedy bee f405 v 3 using bluejay firmware on ESC I want to enable dshot on it but whenever I clicked reboot and goes to multishot so is there any possible way to use  gps with compass on speedy bee. Also tried Inav 7 it works on it but whenever I tried to arm the quadcopter it just flips towards the left side and stick moments are working in the receiver tab but when I arm and try to control there are no stick moments???

DzikuVx, 2023-11-16:
Install INAV 7
https://github.com/iNavFlight/inav-configurator/releases/tag/7.0.0-RC1
https://github.com/iNavFlight/inav/releases/tag/7.0.0-RC2

Diywithall, 2023-11-16:
Tried Inav 7 the dshot is working perfectly but the issue is in receiver tab  receiving pitch roll and throttle  values but when I arm the quadcopter the pitch roll doesn't work and yaw does not work  only throttle works and then flips to a right side

Helipro, 2023-11-17:
> Tried Inav 7 the dshot is working perfectly but the issue is in receiver tab receiving pitch roll and throttle values but when I arm the quadcopter the pitch roll doesn't work and yaw does not work only throttle works and then flips to a right side

Are the motors setup correctly and props spinning the correct direction?  Is the board alignment correct?