iNavFlight/inav #9492: Rangefinder VL53L1X with -1 output

fraro, 2023-11-15:
## Current Behavior
I have a standard 5" quad with Matek F405STD board. VL53L1X connected via I2C is detected and status OK. Sensor output in configurator always shows -1 cm. Same setup works great with a VL53L0X sensor, which shows exact distance.
My VL53L1X sensor is verified to work as expected on a Pi Pico and other setups.

## Steps to Reproduce
1. Connect VL53L1X sensor to board, set VL53L1X as rangefinder in configuration
2. Power up, wait for Sonar = OK
3. In configurator, go to Sensors and check values for Sonar
4. Move hand in front of VL53L1X to watch changes (in my case: no change, always -1)

## Expected behavior
Have the range correctly detected. 

## Suggested solution(s)
Looking for help where to start debugging.

## Additional context
[image]
[image]

CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883, RANGEFINDER=VL53L1X, PITOT=VIRTUAL
...
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=OK, OPFLOW=NONE, GPS=OK
....
Stack size: 6144, Stack address: 0x10010000, Heap available: 1628
I2C Errors: 0, config size: 9543, max available config: 131072

I tried to lower I2C speed from 800 to 400, but no change.

---
# version
# INAV/MATEKF405 7.0.0 Nov  8 2023 / 14:58:44 (a54849be)
# GCC-10.3.1 20210824 (release)
(does not work in 6.x versions either, did not test older versions)

DzikuVx, 2023-11-21:
`-1` mean out of reach or tilt angle is too high. AGL works only when pointed down. Screenshot just taken from my setup

[image]

closed 2023-11-21 by DzikuVx

fraroas, 2023-11-21:
It works perfect when I replace the module by the shorter range VL53L0X version. But I would like to use a VL53L1X, and this does not work for me. I'm totally aware of these ToF module characteristics, I'm using them on several projects.

Are the errors
#define RANGEFINDER_OUT_OF_RANGE        (-1)
#define RANGEFINDER_HARDWARE_FAILURE    (-2)
#define RANGEFINDER_NO_NEW_DATA         (-3)
directly passed to the sensors graph as numbers??
So a "no new data" would give a -3 in the graph?
(with -1 in the graph means at least correct readings, but out of range, but no i2c or reading error) 
I will compare the detection and calibration settings with the arduino/python code I'm using on other stuff.

Thanks!
Frank