iNavFlight/inav #9497: Custom OSD elements driven by programing tab

error414, 2023-11-17:
## Current Behavior
not in inav

## Desired Behavior
Add custom element driven by programming tab

## Suggested Solution
I created proof of concept of custom OSD elements. Element is divided to three part and each part can be set to a single option
* **none**   don't use that part
* **text**      show text (for save memory can be set as "text" only one part of element)    
* **icon static** user select icon which he wants to show
* **icon gv**  icon is driven by global variable value
* **gv**     shows global variable value

GUI is in develop, I just needed to be able to load/save values 
[image]

**the settings on sreen is:** 
icon driven by global variable 0 | text (test) | gloval variable 7 value | visibility driven by global variable 2

A simple program for testing
[image]
Desription of program:   
* there is counter 0 - 30
* in time 0- 10 is showed icon 208
* in time 10 - 20 is showed icon 210
* in time 20 - 30 is element hidden

How it runs on HITL. https://www.youtube.com/watch?v=RtlrOBoGQXU

## Who does this impact? Who is this for?
Pawel introduced auto RTH if plane reach some distance. It's was written in inav programming. Custom OSD elements can provide extra information for pilot what is happening.  So far inav programming does not have any fall back how to inform pilot.  With inav programming + custom OSD elements can be replaced for example "Switch Indicator Settings" or any current simple OSD element. 

So main question is. **Do you want it in INAV?**  I can continue of work, I'm going to make configurator more friendly to user, and code for INAV make more pretty.

Jetrell, 2023-11-17:
I've often thought about something similar to this.
In the past. Systems like the Eagle Tree Vector. Had the ability for the user to write custom text fields or warnings on the OSD to suit their needs.
It is possible in the programming tab, to write rather complex functions.  I think it would be useful to have the flexibility to write a message stating the function, and its state of operation to the user.

An example of this. Was a guy who wrote a fixedwing cruise mode based on airspeed, using the programming tab.
It would be useful for him to display the function and the speed setpoint etc.
This is just one case. The possibilities could be endless.

error414, 2023-11-18:
Even MWOSD has possibility to write custom messages. 

I ignored programming tab in inav for a long time because I didn't see purpose of that, Pawel talked about programming tab in his video on YT, then I started to make sense for me. 

Curretly inav contains posibility to show global variable value in OSD, but it's just for exmaple "GV 155" GV + value. it can be used for debuging program but it does not look good for end users, so it's a reasson a created custom OSD element. You are right possibilities are endless :). 

For 3 custom elements, and text size 10 chars new functionality raises hex size by 1.7KB

RomanLut, 2023-11-19:
Another use case:
https://github.com/iNavFlight/inav/pull/7908

ShikOfTheRa, 2023-11-21:
It is quite cool. I use same functionality in MWOSD plane to display FLAPS and other. Also for target icons on OSD.

closed 2024-02-18 by error414