iNavFlight/inav #9500: In a fixed fing mission, loaded from checked OK, P1 value in a waypoint is checked OK not ignored in a fixed wing (as expected) but translates in a position hold time at the waypoint

Torchi, 2023-11-19:
Load a mission file (that was used in a multirotor) on a fixed wing.   It checks ok and mission is flown.

In the fixed wing the WAYPOINT P1 is useless and should be ignored... instead it is interpreted as a value in seconds to hold the waypoint position.  

When the mission file will be used in a wing, all WAYPOINT P1 should be edited to zero.   I nevertheless think that, in a  wing, WAYPOINT P1 should be just ignored (or rejected at time of mission checks).

Note: this issue may be related to  #9052.

Example of behaviour:

[wing.mission.txt](https://github.com/iNavFlight/inav/files/13402640/wing.mission.txt)

[image]

MrD-RC, 2023-11-19:
The mission is being flown as requested. If you don’t want it to loiter for that time limit, you should change the mission. There are times when pilots do want to use this feature. It is working as expected.

stronnag, 2023-11-19:
The mission is NOT being flown as designed or requested.
* WAYPOINT P1 i used for MR speed and should be ignored for FW
* POSHOLD_TIME P1 is used for loiter time.

In the example mission, the WPs are WAYPOINT, so P1 should not be used for loiter time.

stronnag, 2023-11-19:
https://github.com/iNavFlight/inav/wiki/MSP-Navigation-Messages

breadoven, 2023-11-20:
Yes it shouldn't do this and looking at the code it's not obvious why it's happening. @Torchi do you have `nav_wp_enforce_altitude `turned ON ? It should only do a timed hold if it's a POSHOLD_TIME WP regardless of multirotor or fixed wing. `nav_wp_enforce_altitude `uses the same routine as timed hold to hold position when adjusting altitude but when the altitude is reached it specifically moves on to the next WP if the current WP is a basic WAYPOINT. It shouldn't do a timed hold. I think this was actually fixed before so odd that it's still doing it.

breadoven, 2023-11-20:
OK. Will need to do some testing to work out why it's happening.

breadoven, 2023-11-20:
I tested on HITL with a similar mission, all basic WPs with P1 set to 1000, altitude enforce set to 500 ... and it doesn't do a timed hold at any of the WPs. It adjusted some of the WPs for altitude but no timed hold. It's hard to see how it could looking at the code so not sure what's going here I'm afraid. Did it do it at all the WPs or perhaps only after the Jump WP ?

Torchi, 2023-11-20:
I have forgotten to say that the same mission (then with the same useless '1000' P1), same wing etc., I tested  the week earlier  but then the FC was still loaded with INAV 2.6.1 :  all the WPs of  the mission were flown.   Final JUMP was not executed ?!   so the mission was flown only once and not twice as programmed. The wing finally remained loitering around last WP.  
I thought that the 'problem' was due to the early INAV version and that was one of the reasons I update it.

breadoven, 2023-11-20:
Should work in the simulator the same as for real ... certainly as far as control logic is concerned.

I tested again with a final Jump WP restarting at WP 1 and still no timed hold so I don't know what the issue is. This is testing with version 7.0 but nothing's changed that I'm aware of between 6.1.1 and 7.0 regarding WP Hold Time so both versions should behave the same.

Torchi, 2023-11-26:
Yesterday I made two flights, to double check this issue.  I  confirm it. 

I used a model different from last week one, loaded with INAV 6.1.1.  The first mission loaded had just two corrections from mission last week: P1 parameter was corrected to zero and flight altitude increased (cautiously) to 60m.  The mission was executed fully.  The double eight was not "pretty" and probably would require tuning the parameters related to missions (and btw yesterday's there was a strong wind).  Landed, I changed the battery and I loaded last week's mission: P1 parameter set to 1000 and 50m flight altitude. Result: the mission was stuck at the first WP waiting probably full 1000s.
I attach a zip containing for both flights: mission file and BB (including GPS data, can be replied).

[mission wing test.zip](https://github.com/iNavFlight/inav/files/13466275/mission.wing.test.zip)

breadoven, 2023-11-26:
OK I've worked out the issue here ... the change given in #9052 wasn't included in 6.1.1. Not sure why given it was merged into the master in May whereas 6.1.1 seems to have been created from master in June. Possibly it's because #9052 wasn't flagged with a 6.1.1 milestone so wasn't included in that release. Either way this issue is fixed in 7.0.0 so this can be closed now.

closed 2023-11-26 by breadoven