iNavFlight/inav #9504: RTH/failsafe throttle for fixed wing

P-I-Engineer, 2023-11-20:
## Current Behavior
cruise throttle is used

## Desired Behavior
a set throttle value for failsafe

## Suggested Solution
add a parameter for setting rth throttle. currently if you have your cruise throttle low for maximum efficiency/flight time. during a failsafe you'll be crawling back home. I'd prefer to have it ramp to higher throttle.

for example:
cruise throttle: 1350
rth throttle: 1700

I will implement this in the programming tab for now, i'm throwing out the suggestion to help others.

MrD-RC, 2023-11-20:
Cruise throttle is the correct throttle for auto throttle based navigation modes. Cruise throttle should not be set to the lowest throttle you can get away with. That actually isn’t the most efficient speed to fly at. It will give most flight time, but not cover the most ground. You want the Carson speed for the most efficient ground coverage. 

Also, depending on when FS/RTH activities. You may need to be at an efficient throttle level. Having a cruise throttle of 1350 and then RTH at 1700 may very well mean that you won’t make it back at all because of remaining battery capacity. 

Why is it you want this faster RTH throttle? Just to return quicker? That’s all it will achieve. But does run thr risk of not returning at all.

MrD-RC, 2023-11-20:
Maybe stick to the programming then. I think it's better to keep things consistent. Some pilots may not consider the potential battery issues when flying back at a higher throttle.

P-I-Engineer, 2023-11-20:
sounds good, thanks for the comments