iNavFlight/inav #9511: INAV goes into failsafe after pressing 'SAVE' in the configurator UI

joosttromp, 2023-11-24:
## Current Behavior
I'm running:
Matek H743 Wing V3 on INAV 7.0.0 RC3
BetaFPV SuperD on ELRS 3.3.0
333Hz full - 12ch Mixed
INAV Configurator 7.0.0 RC2
Radiomaster TX16 & BetaFPV SuperG module

in a:
E-flite Viper 90
Flycolor X-Cross HV3-160A (G071) on BLHeli_32 v32.9
BLHeliSuite32 v32.10.0.0
Connected to S1 at DShot600 and Telemetry to RX8.
Freewing 6S 4068-1750KV EDF
Spektrum 6S 5000mAh 50C

After pressing 'SAVE' in the configurator UI, the RX goes into failsafe, where previously there was a connection with the TX. This only happens when the FC is powered by the battery. When the battery is not connected, so the FC is only connected by USB, this does not happen. At the moment the motor beeps, confirming the save, the FC goes into failsafe.

## Steps to Reproduce
1. Connect Battery to ESC
2. Connect USB to FC and connect Configurator UI
3. Go to any tab that has a 'SAVE' button and press it (for example in OSD tab)
4. Once the motor beeps, confirming the save, the FC/RX goes into failsafe and connection with the TX is lost.

## Expected behavior
That the FC/RX does not go into failsafe when a battery is connected and save is pressed in the UI.
When no battery is connected, the RX does not go into failsafe after pressing 'SAVE', but then the DJI air unit, servo's etc. are also not powered.

## Suggested solution(s)
Updating the ESC's BLHeli_32 firmware to v32.10 initially seemed to have resolved this issue, but after further testing, the exact same symptoms seem have have returned.

Just after flashing the ESC, after a save, the motor beeped THREE TIMES (low-low-high), and did not enter failsafe mode, however it did lose connection with the TX, and after repeating the tests, the exact same behavior as previously described occurred.

## Additional context
- INAV version string:
# INAV/MATEKH743 7.0.0 Nov 19 2023 / 19:54:32 (2e9b1c7d)
# GCC-10.3.1 20210824 (release)

rts18, 2023-11-24:
When save is pressed data is written to the eeprom.
It is expected behavior to observe the RX link drop temporarily, along with other functions when data is saved.

>At the moment the motor beeps, confirming the save, the FC goes into failsafe.

The ESC beep isn't confirming you saved the settings. It too is rebooting after the settings are saved to the eeprom.
Settings can not be saved when the craft is in flight. So you need not be concerned. You will not loose the RX link in flight.

joosttromp, 2023-11-24:
Thanks for that information!
However, the RX signal does not come back after an ESC reboot.
Only after the entire flight controller is reboot, the RX signal comes back.

Jetrell, 2023-11-28:
> However, the RX signal does not come back after an ESC reboot. Only after the entire flight controller is reboot, the RX signal comes back.

I was helping a friend setup a 7.0 build with Immersion Ghost. And it did the same thing you reported.
 But I haven't experienced this behavior with CRSF.. Which seems odd. Because you're using ELRS.
As the other guys said. It won't cause any problems in flight.

b14ckyy, 2024-03-22:
Can you please check if it still happens with 7.0 final ir 7.1 RC1? Or is this solved?

Jetrell, 2024-03-22:
@b14ckyy This has not been addressed yet.

b14ckyy, 2024-03-22:
Ah I was not sure if it was even reproducible.