iNavFlight/inav #9518: NEUTRONRC F435 Mini AIO

Guzgog, 2023-11-28:
The new target "NEUTRONRC F435 Mini AIO" does not appear in the Firmware flasher list.

**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

mmosca, 2024-01-09:
INAV only has NEUTRONRCF435MINI target.

I am not sure if the AIO is compatible with this target or not.

jonathanoeijoeng, 2024-01-25:
> The new target "NEUTRONRC F435 Mini AIO" does not appear in the Firmware flasher list.
> 
> **PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**
> 
> * [INAV Discord Server](https://discord.gg/peg2hhbYwN)
> * [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
> * [INAV Official on Telegram](https://t.me/INAVFlight)
> 
> **Please double-check that nobody reported the issue before by using search in this bug tracker.**
> 
> **PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**
> 
> ## Current Behavior
> ## Steps to Reproduce
> 
> ## Expected behavior
> ## Suggested solution(s)
> ## Additional context
> * FC Board name and vendor:
> * INAV version string:

have you tried using F435 mini target?

Guzgog, 2024-02-02:
I have now tried the NEUTRONRCF435MINI target in Inav 7.0, and can confirm it is working fine with the NEUTRONRCF435MINI AIO board, Test hovered all fine. 
I also had Barro and "Walksnail m10" Mag and GPS  and Telemetry working fine.
Can I just say thank you to everyone who has participated. Good work.

closed 2024-02-02 by Guzgog

Kazukifpv, 2025-08-04:
> I have now tried the NEUTRONRCF435MINI target in Inav 7.0, and can confirm it is working fine with the NEUTRONRCF435MINI AIO board, Test hovered all fine. I also had Barro and "Walksnail m10" Mag [...]

I’m confused, you literally said you have used this AIO?