iNavFlight/inav #9520: Integration of Flywoo Goku HD 745 EVOI

PhilippRamjoue, 2023-11-29:
## Current Behavior
I use the Flywoo Goku HD 745 Evo Fligthcontroller (https://flywoo.net/products/goku-hd745-evo-40a-aio-bl_32-dji-plus-play). It's very similar to the Flywoo Goku GN 745. This FC seems to be supported but not the HD 745 EVO. IMU is detected but not the onboard Barometer. Unfortunately without Baro most of the Flightmodes are not possible.

I tried:
- INAV 3.0 (that was supported by Flywoo)
- INAV 5.0 
- INAV 6.0 

I tested it with two different HD 745 EVOs from different orders. Both times Baro (BMP280) was unavailable.

## Desired Behavior
Flywoo Goku HD 745 Evo is fully integrated in INAV because it's a great FC.

## Suggested Solution
?

## Who does this impact? Who is this for?
Due it is a F7 series FC and bigger than the "normal" F745 is great for beginners. There is more space for soldering and with it's size, weight and features is great for nearly every application.

## Additional context
Thx a lot. If I can help or test any changes please reach out to me.

mmosca, 2023-12-04:
Please have a loot at the new hardware policy for INAV. That describes what is needed to add an official target to INAV.

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

closed 2024-01-09 by mmosca

Minutedrone, 2024-07-09:
@mmosca No, I'm not affiliated with flywoo. I have contacted them about this. We have exchanged several times to get them working, but without success. All our crontrollers have the same problem on Inav - no baro.

PhilippRamjoue, 2024-07-09:
The interesting thing is that Arducopter works. I tried several other flight softwares but the only one with that the drone flew was Arducopter. I’m not sure if the real Baro was detected or it was a kind of generic but it worked. My wish is also, that iNav will support. I really like it and arducopter is too nerdy for me.

Minutedrone, 2024-07-09:
Hello,

Thanks but we are not interested.

Cordialement,

L'équipe Minutedrone
--------
*Minutedrone*
*Tel Support : *06 20 35 87 43  //   *Email : ***@***.***
*Site :* https://www.minutedrone.com

 [image: linkedin_circle_1d1d1d.png]
<https://www.linkedin.com/company/minutedrone>   [image:
twitter_circle_1d1d1d.png] <https://www.twitter.com/minutedrone>   [image:
facebook_circle_1d1d1d.png] <https://www.facebook.com/minutedrone/>   [image:
youtube_circle_1d1d1d.png]
<https://www.youtube.com/channel/UC5hg9NTMbLj58EP40JFFC3g>   [image:
instagram_circle_1d1d1d.png] <https://www.instagram.com/minutedrone/>

Minutedrone, 2024-07-16:
Hi !

Good news : 

I had an answer from flywoo after reporting again this issue. 
They sent me a custom firmware version that they made for this flight controller. 
Everything works fine and I got all sensors working fine with inav. 

This is confusing because on their website, the recommanded firmware for flashing is FLYWOOF7DUAL and this firmware doesn't work (creating issues reported here). The firmware file they sent me is FLYWOOF722 and this target isn't available in official inav's official firmware flashing section.

It's also based on an old version of inav  (6.0.0). 

I'm sharing it with you hoping someone can update it to current version of inav and publish this target in the official inav's firmware flashing section so everyone using this flight controler get this problem solved.

Hope this will help.

[inav_6.0.0_FLYWOOF722.hex.zip](https://github.com/user-attachments/files/16251032/inav_6.0.0_FLYWOOF722.hex.zip)