iNavFlight/inav #9524: MSP2 over Crsf Telemety for PID lua Script like betaflight

druckgott, 2023-12-01:
## Current Behavior
MSP2 is currently not Supported over CRSF Telemetry so we cant use lua script to change pids with transmitter

## Desired Behavior
MSP2 is supported over CRSF Telemetry so that we can use a first script and implement more and more things like on betaflight lua script

## Suggested Solution
It would be cool to have a Script like in betaflight to send PIDS and so on with LUA to change it on the Transmitter and do not need a App ore PC or anything else.

[klutvott123](https://github.com/klutvott123) have thxfully implementet mps2 for the lua script
https://github.com/betaflight/betaflight-tx-lua-scripts/issues/491

I have done a version which already starts with inav but only gets msp1 values which is currently not very usefull for PIDS.
https://github.com/druckgott/inav-tx-lua-scripts/
I will later make a fork from betaflight script and change again all so that the commit history from betaflight is also availibe.

## Who does this impact? Who is this for?
Everybody who will use a lua script on Transmitter to change PIds without extra tool and not with any switches in INV which maybe not availble due to other functions

## Additional context
Would be realy cool if we can do this and then extend the lua script lick the betaflight lua script

druckgott, 2023-12-02:
Did Start here no Compile error alrady, but Lua Script says Waiting for API and I do not know how to debug both (lua and inav itself)
https://github.com/iNavFlight/inav/pull/9528

MrD-RC, 2023-12-03:
Honestly, I find the adjustments so much more convenient. You can do it in flight. No need to land, remove your goggles, and look at your radio.

druckgott, 2023-12-03:
It depends how you want to do it. For me in flight is not realy nice, I have do use so much buttons have to concentrate. I think this is depending on the people how they want to do it. Maybe for you it´s not nice but I think there are other people who find it nice. Maybe lets focus on adding good tings to inav ;-). I don´t want to discuss the preferences from people ;-).

RomanLut, 2023-12-13:
No so excited about tuning pids with lua script on my Jumper T-Lite screen :smiley:, but this feature would make wireless configurator communication possible https://github.com/iNavFlight/inav-configurator/issues/1616