iNavFlight/inav #9530: Configuração do modo AIRPLANE no omnibus f4 v3s plus

HigorJales, 2023-12-02:
Olá, estou tentando configurar minha controladora Omnibus F4 v3s plus no modo AIRPLANE mas não estou conseguindo. Consegui configurar o receptor e o mesmo já esta enviando sinal para a controladora e sendo reconhecido, mas não consigo enviar os comandos para os atuadores como motor e servos. 
Radio utilizado:  Futaba T8J
Receptor: Futaba R2008SB

mmosca, 2023-12-04:
Oi Higor, você pode tentar procurar ajuda no discord do INAV, mas em geral usamos inglês aqui e no Discord.

Você chegou a ligar a saída de servos e motores na tab de outputs?

HigorJales, 2023-12-04:
Olá Marcelo. Desde ja obrigado pelas dicas. 
" Você chegou a ligar a saída de servos e motores na aba de saídas?"
Sim, já deixei tudo conectado. Atualmente através do betaflight consigo ver que a controladora esta recebendo os sinais do Radio controle. Já configurei o modo "Arm", quando aciono a chave a controladora reconhece e faz os beebs de aviso mas ao tentar acionar o motor ou os servos nada acontece. Se eu for na aba "motores" no betaflight e ir em "modo de teste de motor\armar" e realizar por la o comando o motor funciona.

mmosca, 2023-12-05:
Você está usando Betaflight ou INAV? Se for Betaflight, você está no projeto errado. INAV é outro projeto.

HigorJales, 2023-12-06:
Marcelo, minha FC é a Omnibus F4 V3S Plus, sabe me informar se consigo configurá-la pelo INAV?

0crap, 2023-12-09:
Please use English language.

closed 2024-03-22 by b14ckyy

felipe12yes, 2024-08-03:
Can someone help me? I'm trying to configure the magnetometer in the inav but I can't. My controller is FC, BetaFlight F4 V3S Plus, I'm using a GPS GNSS Module for GLONASS, BN-880

sensei-hacker, 2024-08-03:
> Can someone help me? I'm trying to configure the magnetometer in the inav but I can't. My controller is FC, BetaFlight F4 V3S Plus, I'm using a GPS GNSS Module for GLONASS, BN-880

 Please open your own issue. Describe what you are trying the accomplish and what you are seeing instead. For example, from your current description we don't know whether you're asking for a link to the docs because you have no idea how to connect a mag, or maybe calibrate it, or you're getting incorrect readings, or maybe an error message, or ... 

Provide screenshots of what you configured, along with preferably a photo of your wiring. 
Note Airbot Omnibus never made a version named "V3S Plus", so what you're dealing with is random unknown manufacturer trying to pass off whatever as if it were an Omnibus.  That means we don't really know what we're dealing with - maybe this board doesn't even have i2c pads to connect a compass at all.