iNavFlight/inav #9547: Iflight F7 TWING uart and motor output conflict

JustNoLimit, 2023-12-07:
## Current Behavior

https://github.com/iNavFlight/inav/assets/33693869/44a93b64-b004-4bda-b9bb-32caa9a81b6d

## Steps to Reproduce
1. If any device is connected to uart 4
2. If any device is connected to uart 5

## Expected behavior
GPS doesn’t work on Uart4 or Uart5. Conflicts with M1 servo output. GPS works on Uart2. RX works on Uart1. Servo outputs working on M1 - M4. I know that uart 3 is used for scl and sda but cant use uart 4 and 5.
## Suggested solution(s)
I think there is something wrong with output mapping

## Additional context
https://pastebin.com/MDszQVYQ

---

- FC Board name and vendor: Vendor Iflight, board name F7 TwingG BT
- INAV version string: 
# version
# INAV/IFLIGHTF7_TWING 7.0.0 Dec  5 2023 / 10:05:11 (895a4f31)
# GCC-10.3.1 20210824 (release)

b14ckyy, 2024-03-22:
checked the target file and I see no assignment conflict on first glance. 
can you try with 7.1 and 6.1.1 if anything changes? 
anyone has the same FC and can validate? Maybe a hardware fault.

b14ckyy, 2026-09-22:
Closed as inactive

closed 2026-09-22 by b14ckyy