iNavFlight/inav #9560: INAV 7.00 MAMBAF722_X8 Barometer (SPL006) UNAVAILABLE

Lima-fpv, 2023-12-11:
## Current Behavior
INAV 7.0 flashed on the Mamba F722 MK4 App, Barometer (tested all) should be SPL006 even if choosen the Barometer in CLI "status" it says Unavailable
Side note: if Baro is selected and configuration tab saved, the log entry is:
"2023-12-11 @ 23:23:09 -- You need to upgrade your firmware before you can use the Configuration tab."
latest INAV configurator and latest INAV stable installed

## Steps to Reproduce
1. Clean Flash INAV 7.0
2. Choose Barometer AUTO > after reboot Baro = NONE
3. Chose Barometer SPL006 > CLI "status" Barometer = UNAVAILABLE

## Expected behavior
Have a Barometer ?

## Suggested solution(s)
Workaround to get the Barometer working ? Maybe PIN configuration possible ?

## Additional context
Dump file of CLI (Baro set to NONE)
(https://pastebin.com/PhT4aFHp)
Dump file of CLI (Baro set to SPL006)
https://pastebin.com/X2zzaRU4

---
- FC Board name and vendor:
Mamba F722 MK4 App
- INAV version string:
# version
# INAV/MAMBAF722_X8 7.0.0 Dec  5 2023 / 10:24:57 (895a4f31)
# GCC-10.3.1 20210824 (release)

Jetrell, 2023-12-12:
You didn't state if this was the first firmware loaded onto this flight controller, from when you purchased it.
Do you have the flight controller power by a battery when connected to the Configurator ?
In my experience with Mamba flight controllers. The barometer is not powered by the USB supply.

Lima-fpv, 2023-12-12:
> You didn't state if this was the first firmware loaded onto this flight controller, from when you purchased it. Do you have the flight controller power by a battery when connected to the Configurator ? In my experience with Mamba flight controllers. The barometer is not powered by the USB supply.

I had betaflight first on the fc, but I didn’t used the barometer-however I tried with and without battery. I will try later again / dumb question: the barometer should work inside my Appartement or just outside ?

Lima-fpv, 2023-12-14:
status
INAV/MAMBAF722_X8 7.0.0 Dec  5 2023 / 10:24:57 (895a4f31)
GCC-10.3.1 20210824 (release)
System Uptime: 16 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 24.74V (6S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=UNAVAILABLE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x20010000, Heap available: 1716
I2C Errors: 1081, config size: 9401, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 2, used = none
   CURRENT : configured = ADC 3, used = ADC 3
  AIRSPEED : configured = none, used = none
System load: 40, cycle time: 505, PID rate: 1980, RX rate: 239, System rate: 9
Arming disabled flags: NAV HWFAIL CLI
OSD: MSP DisplayPort: BetaFlight Compatability mode (HD) [53 x 20]
VTX: not detected
GPS: HW Version: UBLOX10 Proto: 34.09 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/1
    BeiDou 0/1
    Glonass 0/0
    Max concurrent: 3

so with Battery and SPL006 choosen, still Baro unavailable

Lima-fpv, 2023-12-16:
topic closed - had to choose another target for flashing the firmware ...

closed 2023-12-16 by Lima-fpv