iNavFlight/inav #9569: 3D flying - hovering - torque roll - holding flight attitude

DM4DS, 2023-12-15:
## Current Behavior
As Wing pilot, flying "3D" in AGRO is nearly impossible, so I switch back to manual mode to have 3D fun. The FC is doing it´s job in AGRO, but there should be a "3D wing" mode or so. Don´t know how to call it correct. Also for the new pilots it would be great if the FC could help them, starting 3D flying.

## Desired Behavior
In 3D Wing mode, the FC should be stabilizing but should allow the hover or even torque roll. In knife edge to fly a "death slide". Keeping the plane in its Flight attitude, if there are no RC inputs.

## Suggested Solution
Insert a mode were the current angles (Flight attitude) are tryed to hold, also on prop up (torque roll) or knife edge angles.
This works for "normal" flying/situations in AGRO beautiful, but (for me) not in 3D manuveurs. or my settings are wrong.
If the level of "holding angles" could be adjusted via a RC channel, it would be perfect. I mean , activate the mode via switch + a specific RC channel to adjust the % the FC is trying to hold the plane in its current Flight attitude .

There are stand alone gyros, so I am sure you are able to add it into INAV as well.

## Who does this impact? Who is this for?
Wing / Airplane pilots of all sizes and styles. Even simple flying wings are fun to fly this way. It´s just the next level of flying an RC plane.

## Additional context
I fly 3D for over 20 years without FC, but the modern pilot is using FCs as helper, fall back, life safer.
New pilots see us flying in 3D and wanna do the same and to make it easier to learn, the FC should help them. Adjustable, so by become better and better, they can turn down the % the FC is helping them.

If this is possible in INAV 7 as it is, just my settings are wrong, would be great to tell me :)

Thanks for your amazing work on the firmware, software.
fly safe
Sandor - DM4DS

Jetrell, 2023-12-15:
Since you have been flying 3D for many years.. You would know that most airplanes are not designed to perform 3D flight. 
What you're requesting isn't an easy thing to accomplish. There are too many requirements for such a mode, from what is primarily FPV flight software.
And it would also require a high level of tuning by the user, to suit the specific model they are using.

There is a new mode in the next 7.1.0 release. Called ANGLE HOLD.. Which will attempt to hold attitude on the pitch and roll axis's. But it isn't ideal for 3D flight.

b14ckyy, 2024-03-22:
did you try manual tuning the PIDs and work with very high Integral values and less Feed forward? Integral is basically what holds attittude in Acro (The plane should not be aggro :P) 

but the plane of course still needs to be aerodynamically capable to do these maneuvers in manual before the PIDFF controller can assist. Tuning yaw can be challenging though.