iNavFlight/inav #9573: no return video with HDzero and SPEEDYBEEF405V3

lasdevegas, 2023-12-16:
## Current Behavior

INAV 7.0 just flashed on the SPEEDYBEEF405V3 with inav 7 configurator , no diff applied .
Turn on my googles HDzero and return video is fine .
MSP display port activate on port 3 , no more return video
The same configuration is fine with inav 6 and betaflight.

## Steps to Reproduce
Clean Flash INAV 7.0
Choose MSP display port for hdzero OSD
return video is ko 

## Expected behavior
return video ok

## Suggested solution(s)
works fine with inav 6 
different hdzero port configuration between inav 6 and inav 7  ?

## Additional context

https://pastebin.com/zfAfqrht

# INAV/SPEEDYBEEF405V3 7.0.0 Dec  5 2023 / 11:05:24 (895a4f31)
# GCC-10.3.1 20210824 (release)

aerostarlegacy, 2023-12-22:
Hello, I am also having this issue with Matek F405-WMN and inav 7. Currently checking if inav 6.1 works. Will update.

aviphysics, 2023-12-28:
I am having this issue and 
have been able to replicate on a
* Matek F405 MiniTE 
* Holybro KakuteH7Mini

VTX in a HDZero Freestyle V2 running Rev 13112023 (aka v1.5.0)

Not a practical solution, but I can get video with OSD temporarily by
1) connect FC with USB to PC
2) connect to FC with configurator
3) In configurator disable MSP displayport and press save/restart button
4) power VTX
5) In configurator select MSP displayport on the required uart and press save/restart button
* The VTX will now transmit with OSD and can be controlled with stick commands.
* MSP VTX commands do not work at this point.
* If the FC is disconnected from USB and battery, then repowered, video will not work.

aviphysics, 2023-12-28:
I reflashed the VTX firmware, and now it is working with the Holybro KakuteH7Mini

The Matek F405 MiniTE only works if I power the flight controller up before plugging the VTX into the Flight Controller.

closed 2024-02-02 by DzikuVx