iNavFlight/inav #9585: Matek H743-WLITE flight controller crashes and stalls when EEPROM is saved without reboot on V7.0.0(stable)

chucky-whiskers, 2023-12-21:
**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

INAV runs properly on flight controller when configurator is connected until EEPROM is saved without a reboot. Radio connection is still present, but the servos and motors lock up and receive no commands (even in manual mode). Configurator also can no longer communicate with FC until rebooted.

## Steps to Reproduce

1. Flash INAV v7.0.0 (latest stable) to Matek H743-WLITE
2. Configure any setting saved to EEPROM 
3. Save without reboot
4. Try to move control surfaces

## Expected behavior

The EEPROM settings should save and the program should continue to run without glitches. 
## Suggested solution(s)

Force a reboot when saving all settings or find the glitch in the program. I'm worried that something like auto-tune will save in-flight and cause the system (and plane) to crash.
## Additional context

I HAVE TRIED REFLASHING FIRMWARE.

https://pastebin.com/g2aArQcd
---

- FC Board name and vendor: 
Matek H743-WLITE Amazon
- INAV version string: 
INAV/MATEKH743 7.0.0 Dec  5 2023 / 10:42:54 (895a4f31)
GCC-10.3.1 20210824 (RELEASE)

MrD-RC, 2023-12-21:
Not a solution. But hopefully it will put your mind at ease. No settings are save in flight. You have to disarm to trigger that save event. Then it is only continuous servo trim and stats that will do this. 

Auto tune and auto level trim do not call for the settings to be saved.

rmaia3d, 2023-12-25:
Could this be related to the compiler toolchain change for Inav 7 and optimization levels conflict that has already been discussed here?

https://github.com/iNavFlight/inav/issues/8905#issuecomment-1495278006

Jetrell, 2024-01-18:
@chucky-whiskers I agree with your observation.
Do you also experience problems with this flight controller refusing to connect to the configurator, when black box logging is enabled ?
I've found this to be the case with every SD card and workaround I have tried. And nothing short of disabling the logging helps.

b14ckyy, 2024-03-22:
Can you please check if this is still an issue with 7.1 RC1? Then we can have a look again.

Jetrell, 2024-03-22:
Yep. Its still a problem. Especially when logging is enable. IF you're lucky to get it to log in the first place.

b14ckyy, 2024-03-22:
OK testing time tomorrow. Have not yet updated my clouds above 6.1.1

skumor-foreflight, 2024-07-08:
I am having this issue while saving custom programming logic. Once I save receiver input freezes

RickMod, 2024-07-25:
Same issue on Matek H743 Wing V2 Inav 7.1.1 FC firmware 7.1.2
 
Found about the issue during OSD setup. Lucky me, as I had continuus servo trim on, and planned Maiden with fpv, not classic. 2.4m glider wing, ~1.5kgs, flying by itself .... quite scary i would say.

Regarding black box SD card:  it was quite frustrating before i found solution which is: do not to use it :)

b14ckyy, 2025-04-07:
Fixed in 8.0

closed 2025-04-07 by b14ckyy