iNavFlight/inav #9593: Doesn't detect gyro and accell Inav 7.0.0,

Restuadkh, 2023-12-23:
## Current Behavior

## Steps to Reproduce

1. Instal inav 7.0.0 into Target : FURYF4OSD, with accl ICM42668P
[image]
2. set to ICM42605
[image]
3. alternatif i adding the sensor mpu6500 and mpu6050, but is a same
4. 

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

mmosca, 2024-01-09:
What happens if you set accelerometer to auto and reboot?

mmosca, 2024-01-25:
No response.

closed 2024-01-25 by mmosca