iNavFlight/inav #9598: 7.0.0 MAMBAF405 motor output broken

laurin-1, 2023-12-26:
Motor output was working as expected with 6.1.1.

version
INAV/MAMBAF405 7.0.0 Dec  5 2023 / 09:58:09 (895a4f31)
GCC-10.3.1 20210824 (release)

After flashing 7.0.0, when using the outputs tab, when trying to spin motor 2 motor 4 is spun, and when trying to spin motor 4 motor 1 is spun. Nothing happens when trying to spin motor 1 or motor 3. Passtrough in ESC-configurator doesn't work either.

After downgrading to 6.1.1, everything is working as expected again.

I'm unsure what's the cause of this, it might be because of the changes to timer allocation in [https://github.com/iNavFlight/inav/blob/master/src/main/target/FURYF4OSD/target.c](url)
(Previously: TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, Now: TIM_USE_OUTPUT_AUTO)

sensei-hacker, 2023-12-26:
Laurin, can I see a screenshot of your motor mixer, please.

laurin-1, 2023-12-27:
[image]
[image]
The automatic timer allocation seems to only allocate timer 2 for motors in case of my flight controller, after manually changing both timer 2 and timer 3 to motors, everything is working as expected.
I was aware that something was changed in regards to timer allocation and I've read the release notes when INAV 7 initially got released, but I somehow didn't remember that it had to manually be changed.
Thanks a lot for your help!

closed 2023-12-27 by laurin-1