iNavFlight/inav #9600: Uncontrollable motor rev up on SpeedyBee F405 V3 BLS 50A 30x30 FC&ESC Stack on a simple throttle test

johndoe422, 2023-12-28:
Stack: SpeedyBee F405 V3 BLS 50A 30x30 FC&ESC Stack
Form Factor: 5 inch quad, no FPV system. Just 4S batteries, the stack, and the Tx-Rx. 
Spec: 2300kv SunnySky motors, 5 inch 2 blade props, CF frame, Radiomaster TX12 MKII ELRS Ts and Rx.

Config : Reduced throttle scale to 70%, Expo throttle curve (0.65), reduced motor spin limit to 75%. Tried Dshot300 & Dshot600.

Issue: The quad armed alright. After that, I tested the throttle. Just a small nudge of the throttle stick up and quickly down. The expected result was that the motors should spin up and come back to idle speed quickly. 
Observation: When the throttle was increased, the motors spinned up but they kept spinning all the way up even after the throttle stick was brought down to the lowest position quickly. Luckily I had the drone restrained that prevented a flyaway. I disarmed quickly too.

This was reproducible 100% of times. I tried only with DShot300 and Dshot600.

johndoe422, 2023-12-28:
I missed to mention that I had INAV7 firmware. I clean flashed INAV7 onto the stack (Full chip erase) and configured everything manually. Ensured motor order, motor directions etc. and cross verified a couple of times. I had props-in rotation pattern.

sensei-hacker, 2023-12-28:
What flight mode were you in?

johndoe422, 2023-12-28:
I was in Angle mode.

sensei-hacker, 2023-12-28:
Consider you're in angle mode, and the roll / pitch stick is centered. What is that commanding the aircraft to do?  It's commanding the aircraft to level itself, and any payload attached to it.

If it (and the table or floor) are 4° off level, it will try to level itself. If it has a payload attached to it, it will still try to level. Including if that payload is a table.

You can't test flight characteristics without flying.

johndoe422, 2023-12-28:
Thank you for the comment. I understood your point totally and I would like to confirm that it wasn't the case. The craft wasn't trying desperately to level itself after sensing a tilted orientation. I did the exact same test with Betaflight 4.4.3. The craft behaved as expected - when I reduced the throttle to 0, the motors spinned down. In both cases, the craft was placed in a level ground (floor) and there was absolutely no tilt. In fact, I used the same floor to calibrate the horizontal orientation.
Would really like to migrate to INAV for its features like RTH. Cruise and Alt hold etc. I am happy to test further if needed.

sensei-hacker, 2023-12-28:
A good test would be to fly the quad without having a table attached to it. If you're worried keep in mind where the disarm switch is.

> the craft was placed in a level ground (floor) and there was absolutely no tilt.

There's literally no such thing. For $150 million you can get pretty close, while making the primary mirror for the Hubble. Still, shifting gravity ruins your day, as magma flows within the earth. The floor of your house might easily be within 10.0° or so of level - maybe Even 2.0°, but 10.0°  or even 1.0° isn't 0.0°.

Suppose, though, you were in a universe where such a surface existed.  STILL the $3 accelerometer on your FC is going to read a couple degrees off from moment to moment.

johndoe422, 2023-12-28:
I was trying to help out to make INAV7 better, but it seems to me that you are suggesting there is no problem at all.

I assembled and flew a 10-inch quadcopter based on Ardupilot years ago, and now I am building this 5-inch fpv quad as well. I conducted the same test using Betaflight. Therefore, I have a good understanding of what to anticipate when giving a small throttle nudge to a drone placed on level ground. It should not cause the motors to spin uncontrollably. Moreover, I am very familiar with the sensation of the drone attempting to level itself when I arm it while holding it. If I tilt it, and it will attempt to level itself. However, after achieving level, the motors will calm down until it senses another tilt. So, please don't relate the issue to this behavior. 

I know there is no such thing as perfect level. The more money you invest, the more precision you get. Here the fact is that, these $3 accelerometer equipped drone behaved well with another firmware.

However, this issue was not like that at all. The motors simply started spinning at full speed with the slightest throttle blip. To provide an analogy, it would be similar to starting an ICE car, briefly pressing the accelerator pedal, and expecting the engine to rev up and down slightly. But in this case, the engine goes wild and revs up to the maximum limit, even after releasing the accelerator pedal.

MrD-RC, 2023-12-28:
I don’t fly quads on INAV yet. But couldn’t this be:
- the idle throttle is too high?
- airmode being activated due to the scaling?

@jetrell may have an idea?

johndoe422, 2023-12-28:
I had set the idle throttle at 6%, which isn't too high. I use the same setting with Betaflight and it idles just fine, without any tendency to take off. 

It doesn't look like it is Airmode trying to do anything. The problem is that the motor rpms go progressively high, after a nudge on the throttle stick.

> I don’t fly quads on INAV yet. But couldn’t this be:
> 
> * the idle throttle is too high?
> * airmode being activated due to the scaling?
> 
> @Jetrell may have an idea?

sensei-hacker, 2023-12-28:
> I was trying to help out to make INAV7 better, but it seems to me that you are suggesting there is no problem at all.

What I'm suggesting is IF there is some problem related to flying in Angle, the way to get more information about that would be by flying in Angle.

We already know it can't possibly work while the quad is carrying a table. Can it work when the quad isn't carrying a table? We don't know.

Or to extend your analogy:

I changed the car to a tree, engaged cruise control, and ...

Cruise control isn't *supposed* to work while chained to a tree. Angle and RTH aren't supposed to work while chained to a table. So we simply have no idea whether there is a problem or not.

MrD-RC, 2023-12-28:
> I had set the idle throttle at 6%, which isn't too high. I use the same setting with Betaflight and it idles just fine, without any tendency to take off.
>
> It doesn't look like it is Airmode trying [...]

BF and INAV are different firmware. They aren’t the same in many ways. So comparing and using the same settings may not be the best way to go about things.

breadoven, 2023-12-28:
If you nudged the throttle above the airmode throttle threshold trigger point, even briefly, than airmode will become permanently active and you will not be able to get the throttle to return to idle. The motors will appear to rev up as the quad tries to maintain level. It's not going to fly away like this but can become a bit skittish and bouncy as it bumps off the ground.

DzikuVx, 2023-12-28:
Closing as this is expected behavior when Airmode kicks in after THR crosses a threshold.
Additionaly, please never test a drone on a workbench with props attached, it's extremely dangerous

closed 2023-12-28 by DzikuVx