iNavFlight/inav #9601: Speedybee F405 v3 50A ESC - Motors spin up all the way to the max rpm, if throttle is nudged and brought back to lowest position

johndoe422, 2023-12-29:
Stack: Speedybee F405 v3 50A BLSheli ESCs
Form Factor: 5 inch quad, no FPV system, 2300kv motors, 4S 1550mah batteries, Radiomaster TX12 MKII ELRS Transmitter.

Issue: After arming the quad, if the throttle is increased to say 30-35% and brought back to lowest position just to test the throttle, the motors will start spinning all the way up to the max rpm. Disarming is the only way to stop it. If I had propellers attached the drone would have taken off and ascended rapidly for sure.

The rev up is smooth, and the time taken for reaching max rpm is about 3.5 - 4 seconds.

No, It is NOT Airmode in action. Airmode should not rev up the motors **all the way to max rpm** just like that, when the drone is placed on a level floor. I tested the same board with Betaflight 4.4.3, with Airmode always on, and this does not happen.
 
Attaching the full dump of my config. 
[INAV_7.0.0_cli_20231229_131209.txt](https://github.com/iNavFlight/inav/files/13791974/INAV_7.0.0_cli_20231229_131209.txt)

johndoe422, 2023-12-29:
Pasting below some of the throttle related settings that I changed from their defaults.
throttle_scale =  0.750
throttle_idle =  10.000
thr_expo = 60
thr_mid = 47

MrD-RC, 2023-12-29:
Duplicate of https://github.com/iNavFlight/inav/issues/9600

To me this sounds very much like airmode. The activation threshold for airmode is crossing 15% throttle in INAV. So you are definitely triggering that. The quad is also being disturbed by that air. So airmode is trying to stabilise it. 

Though I highly do not recommend testing with props on. Just to prove a point, you can disable the airmode feature, and see if the same thing happens. My guess is that it won’t.

sensei-hacker, 2023-12-29:
Most of the developers have already told you what's happening.  Pawel closed your issue, telling you the same thing.

Please stop spamming the GitHub.

Your drone can't carry a table. Simple as that.

Jetrell, 2023-12-29:
> No, It is NOT Airmode in action. Airmode should not rev up the motors all the way to max rpm just like that, when the drone is placed on a level floor. 

@johndoe422 What makes you think that ?

In multicopter use. Airmode is designed to provide active stabilization (primarily i-term), regardless of throttle position. Once the `airmode_throttle_threshold` value is exceeded. And will remain active until the FC is disarmed.

When you have the quad sitting on the ground. And are not allowing it to move, with Airmode active. The i-term error will accumulate more and more. until the motors max out at full throttle.
This isn't a fault. Its a function of operation. How is the FC to know you are holding the quad on the ground. and preventing it from actively stabilizing.

closed 2023-12-29 by sensei-hacker

ovel1, 2024-01-02:
I had the same issue but with the propellers on. Once the throttle stick was in the middle position, the drone rapidly took off and hit the nearest wall. luckily only 2 props were broken. no issues with AIRMODE off. I have the same speedyBee FC.

MrD-RC, 2024-01-02:
@ovel1 thank you for confirming. It doesn’t happen with airmode disabled. It is expected behaviour of airmode.

ovel1, 2024-01-02:
> @ovel1 thank you for confirming. It doesn’t happen with airmode disabled. It is expected behaviour of airmode.

hitting the wall on the full speed in the uncontrollable flight? not sure such behaviour is expected tbh...