iNavFlight/inav #9605: HGLRCF722 uses Octo X8, motor 5 is not working

ez4fm, 2023-12-30:
## Current Behavior
HGLRCF722 uses Octo X8, motor 5 is not working, and after testing, all wiring is normal.
I have flashed all firmware versions from 6.0.0 to 7.0.0, and this issue has been persistent.

## Expected behavior
All motors are working properly

- FC Board name and vendor:INAV/HGLRCF722 6.1.0 May 11 2023 / 12:21:38 (d6ce6e70)
- INAV version string:GCC-10.2.1 20201103 (release)

mmosca, 2024-01-09:
Can you try multishot?

I suspect there is a DMA conflict with S5.

b14ckyy, 2024-03-22:
Please test 7.0 or 7.1 RC1. 
multiple DMA fixes, also target specific have been done after 6.1

closed 2024-03-22 by b14ckyy

ez4fm, 2024-03-25:
> Can you try multishot?
> 
> I suspect there is a DMA conflict with S5.

Thank you. I've replaced the other controllers