iNavFlight/inav #9607: Alignment tool problem? Or my problem?

Retlsnake, 2023-12-30:
Ahoj,
mám na letadle řídící jednotku nestandartně uloženou dle obrázku, je tak i na skutečném letadle.

Když sem nastavil v iNav Alignment tool rotaci jednotky tak jak je na letadle, a chtěl sem zkontrolovat v iNav Setup jestli se v animaci letadlo vrtí stejně jak oné animaci tak sem zjistil že se sice vrtí správně ale je vzhůru nohoma (animace) vůči skutečnému letadlu.

Co dělám špatně?

Když bych dal jednotku normálně na plocho placato šipkou dopředu a v Alignment tool nechal nuly takže vše normálně tak mě to bude fungovat a i funguje.    

Ale já už ji mám přidělanou na bok letadla, na hřbetu není místo a už mám natahaný i káble.  

Když Alignment tool zalžu a v Alignment tool ji otočím tak se v animaci vrtí správně (letadlo) a obojí se stejným horizontem.

Když to nechám obelhané tak to asi fungovat bude zisky z akcelerometrů budou asi taky správně to ještě ověřím ale mám strach aby mě to neblblo v navigačních modech.
Sice tam kompas nemám a nebude tam takže by to snad nemuselo ani vadit, ale dost mě to štve!!!

Buď to maj i iNav blbě animovany v tom Alignment tool [b]nebo dělám něco blbě já ale co[/b]? 

Zkoušel sem na třech různých jednotkách simulovat a vždy stejný problém.

Prosím poraďte někdo.

Hello,
I have the control unit on the plane stored non-standardly according to the picture, it is the same on the real plane.

When I set the rotation of the unit in the iNav Alignment tool as it is on the plane, and I wanted to check in the iNav Setup if the plane moves in the animation the same way as the animation, I found out that it moves correctly but is upside down (animation) compared to the real plane .

What am I doing wrong?

If I put the unit normally on a flat surface with the forward arrow and leave zeros in the Alignment tool so that everything is normal, it will work for me and it does.

But I already have it attached to the side of the plane, there is no room on the back and I already have the cables stretched out.

When I set the Alignment tool to false and rotate it in the Alignment tool, it moves correctly in the animation (plane) and both have the same horizon.

If I leave it besieged, it will probably work, the gains from the accelerometers will probably also be correct.
Although I don't have a compass there and it won't be there, so maybe it shouldn't even bother me, but it annoys me a lot!!!

Either the iNav is animated badly in the Alignment tool [b]or I'm doing something stupid but what[/b]?

I tried to simulate on three different units and always the same problem.

Please advise someone.

INAV  Configurator 7.0.0 and 7.0.1

FC: Omnibus f4 v2 pro,GF30F405-ICM,Matek 722 STD

Video: [https://www.youtube.com/watch?v=onPRLfn0kvo](url)

**[image]**

sensei-hacker, 2023-12-30:
If the movement is right except upside down, add 180° to roll and pitch.

Retlsnake, 2023-12-30:
I understand...

In that case, there must be an error in the animation in the Alignment tool menu, because in order for the real plane to tilt the same (and with the same horizon) as in the setup menu, the actual position of the FC on the plane is not as shown in the animation in the Alignment tool!!!

It's either my fault somewhere else. I don't understand the Alignment tool correctly.

Or is the error really in the Alignment tool.

Thank you for the quick response. I would like to not only solve the problem but also understand it.

Retlsnake, 2023-12-31:
Only theoretically, if the Alignment tool provided incorrect settings for the compass as well, so when configured for a copter with a compass, it caused very serious problems with navigation, the drone in GPS navigation modes would have to fly away from the place because the compass would provide a rotated course and the drone would have a cyclical error in move the course further and further away from the set point!!!
​

rts18, 2024-01-14:
> Or is the error really in the Alignment tool.

I believe there is an error. And confusion surrounding what `orientation preset` is referring too.
It appears to work as expected for basic alignment of both IMU and compass. 
**e.g.** If the flight control board or GNSS module only require yaw axis rotation, with respect to each other.

But when the install is of a more complex nature. Like the flight controller being flipped, and a yaw axis offset applied to the compass. I have found the alignment tool to give incorrect values. Comparing with the actual alignment required for successful operation.

Retlsnake, 2024-01-15:
So is there really a bug in the INAV alignment tool? Do you think it will be fixed in the next version of the INAV gui? I also asked about it on FB, but the responses are quite lukewarm and no one has yet told me that it is a fault in inav or mine. Not many people probably turn the unit in PITCH and ROLL. Apparently there is no problem with unit rotation in YAW. I was waiting for some INAV developer to comment on this, but still nothing. They probably have a lot of work elsewhere, because INAV is just great.

PS: My English is quite weak, so if it seems to someone that my comments are going around in circles, it may be because of that...